OpenHantek/openhantek #154: Crash on OSX

DanielO, 2018-01-12:
I pulled a36eccc to test building the firmware extractor and now find OpenHantek crashes after selecting my device with..
````
compile shaders
QOpenGLShader::compile(Vertex): ERROR: 0:2: '' :  version '330' is not supported

*** Problematic Vertex shader source code ***

          #version 330
#define lowp
#define mediump
#define highp
#line 2
          in highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          }

***
Failed to compile OpenGL shader programs.
 "ERROR: 0:2: '' :  version '330' is not supported\n"
zsh: segmentation fault  ./openhantek/OpenHantek
````
The stack trace is
````
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   OpenHantek                    	0x000000010dce57bd QOpenGLFunctions_3_3_Core::glViewport(int, int, int, int) + 29 (qopenglfunctions_3_3_core.h:438)
1   OpenHantek                    	0x000000010dce5648 GlScope::resizeGL(int, int) + 72 (glscope.cpp:246)
2   org.qt-project.QtWidgets      	0x000000010e110028 QOpenGLWidget::resizeEvent(QResizeEvent*) + 104
3   org.qt-project.QtWidgets      	0x000000010e0e8960 QWidget::event(QEvent*) + 1024
4   org.qt-project.QtWidgets      	0x000000010e0a7612 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
5   org.qt-project.QtWidgets      	0x000000010e0a8968 QApplication::notify(QObject*, QEvent*) + 360
6   org.qt-project.QtCore         	0x000000010ec74e4f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
7   org.qt-project.QtWidgets      	0x000000010e0e1286 QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) + 278
8   org.qt-project.QtWidgets      	0x000000010e0e759a QWidgetPrivate::show_helper() + 42
9   org.qt-project.QtWidgets      	0x000000010e0e8303 QWidget::setVisible(bool) + 947
10  org.qt-project.QtWidgets      	0x000000010e0e78f4 QWidgetPrivate::showChildren(bool) + 292
11  org.qt-project.QtWidgets      	0x000000010e0e75bc QWidgetPrivate::show_helper() + 76
12  org.qt-project.QtWidgets      	0x000000010e0e8303 QWidget::setVisible(bool) + 947
13  org.qt-project.QtWidgets      	0x000000010e0e78f4 QWidgetPrivate::showChildren(bool) + 292
14  org.qt-project.QtWidgets      	0x000000010e0e75bc QWidgetPrivate::show_helper() + 76
15  org.qt-project.QtWidgets      	0x000000010e0e8303 QWidget::setVisible(bool) + 947
16  OpenHantek                    	0x000000010dd1068f main + 4159 (main.cpp:141)
17  libdyld.dylib                 	0x00007fff72bd3115 start + 1
````

I think MacOSX's GL implementation is very old :(

FWIW this is on a 2013 MBP running High Sierra with an Intel Iris GPU.

davidgraeff, 2018-01-13:
Hm, according to https://support.apple.com/en-us/HT202823 most macs support OpenGL 3.3. The developer website says something about OpenGL 3.2 core profile. Try changing the version string that fails for you in glscope.cpp from "#version 330" to "#version 320".

DanielO, 2018-01-13:
OK, so there is hope :)
I did some googling and added
````
    // Specify an OpenGL 3.3 format using the Core profile.
    // That is, no old-school fixed pipeline functionality
    QGLFormat glFormat;
    glFormat.setVersion( 3, 3 );
    glFormat.setProfile( QGLFormat::CoreProfile ); // Requires >=Qt-4.8.0
    glFormat.setSampleBuffers( true );
````
after "    QApplication openHantekApplication(argc, argv);" in main.cpp but no change..

Similar to if I try 320 as the version (or 220 and 120..)

davidgraeff, 2018-01-13:
Related to #155

davidgraeff, 2018-01-13:
I have fixed the shader version and did some more things. Wait for the Pull Request to be merged and try it again please :)

closed 2018-01-13 by davidgraeff

DanielO, 2018-01-14:
Yep that works, thanks :)

PR #159 (2018-01-13): Use OpenGL 3.2 instead of 3.3 to make MacOSX work again
Fix:
* Move all GL startup calls to GLScope::fixOpenGLversion
* Depending on OpenGL or ES availability initialize the QSurfaceFormat
* Don't crash if the shader compilation failed. Show a nice message instead.
* Use DsoWidget::showEvent to update all kind of UI things. Otherwise 
  for example the zoom view potentially will be "shown" (setVisible), before the DsoWidget
  finished its constructor and we get multiple `initializeGL()` calls.

Optimize:
* Don't write the markers to the GPU each frame update, only
  do this if they actually ch [... cut]