OpenHantek/openhantek #156: DC/AC mixed up

phmarek, 2018-01-13:
With 0e6d658a0519bbe8bde70b8985f1aaafc21329d9, setting CH1 to AC gives me no more data (doesn't trigger anymore?[1]), while using DC gives me the AC picture (centered around 0V).

Ad 1: not sure about triggering - I moved the trigger level around the position where the levels should be, but got no signal anymore.

phmarek, 2018-01-13:
Ah, it looks like the "invert" setting (checkbox) is used inverted ;)

phmarek, 2018-01-21:
The "Invert" checkbox works for me with ba44a820d61bb923cb2192349360ad1e1efbe63f.

But switching to AC just moves the CH1 line to the lower or upper bottom of the scope area, as soon as the CH1 zero line is outside the ±50% vertical position.

phmarek, 2018-01-26:
Perhaps it's the high-DPI notebook display that causes this? Are some calculations done in pixels but others in inch, or something like that, so that upper and lower quarter values get misinterpreted as "out of range"?

What kind of feedback is required from me to diagnose that problem?

davidgraeff, 2018-01-26:
The DSO model, operating system, Qt version.
Actually I'm also waiting from AC mode users for confirmation. Because my DSO does only support DC mode.

phmarek, 2018-01-26:
DSO2250, Debian testing AMD64, libqt5 is 5.9.2-1.

Thanks a lot!

phmarek, 2018-01-29:
Ah, further testing showed some new aspects.
A major

phmarek, 2018-01-29:
Only the second screenshot (got uploaded in a strange order?!) has approximately right data - the lower level of the wave is about 8V, the upper at about 12V. (Confirmed with an older version of OpenHantek.)

phmarek, 2018-03-07:
Perhaps it helps a bit if I mentioned that the USB communication can be easily seen via

```
tshark -i usbmon3 -Y "(usb.capdata[0] > 6) && (usb.endpoint_address.direction == 0)" -T fields -e frame.number -eusb.capdata "$@"
```

Using that I can see this communication:
```
15691   0f:00:77:ec:07:00:87:eb:07:00
15695   0e:00:02:00:ed:ff:00:00
15699   0d:00:01:00
15703   0c:00:03:00:00:00:00:00
15707   0b:00:02:00
15711   07:00:00:00:00:00:00:00
```

When I'm changing the gains, I can hear the relays clicking, but no SETRELAYS message is visible?!

phmarek, 2018-03-08:
Looking around some more I can see references to a `sdGetChannelLevel()` function in the SDK; it seems that this returns some zero level calibration data.
Do you know any more details about that?

Seems that this is not implemented yet.

phmarek, 2018-03-08:
Here's how it's being used: https://sigrok.org/gitweb/?p=libsigrok.git;a=blob;f=src/hardware/hantek-dso/protocol.c;h=c74187ada435f2441e7aa3b88ed926194c9d530b;hb=HEAD#l191

davidgraeff, 2018-03-09:
Oh we do that actually. Can you also test with the openhantek2 branch, please? There is a debug log that shows all send/received messages and also reports if the zero-level calibration data has been received. The method is called "retrieveOffsetCalibrationData" in `src/hantekdso/dsocontrol.cpp".

phmarek, 2018-03-09:
I'm trying to -- not sure whether it's easily possible.

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt53DRender/Qt53DRenderConfig.cmake:27 (message):
  The imported target "Qt5::3DRender" references the file

     "/usr/lib/x86_64-linux-gnu/qt5/plugins/sceneparsers/libassimpsceneimport.so"

  but this file does not exist.  Possible reasons include:
```
and an upgrade isn't that nice, either:
```
# apt-get install qt3d5-dev -t experimental 
...
90 upgraded, 4 newly installed, 106 to remove and 521 not upgraded.
```

Sadly there are quite a few application I need among the "106 to remove"...

phmarek, 2018-03-09:
A few more manual package installations further I get a binary, but also a lot of warnings:

```
openhantek/src/hantekdso/modelspecification.h:143:38: warning: extra ‘;’ [-Wpedantic]
```

phmarek, 2018-03-09:
First observation with 349068029c9e0efdfa953329df3e05b406b45aed: Sample rate is either 0.14Sa/sec, or  (when clicking the down arrow timebase) 100MSa/sec. Clicking down on samplerate I get then 4.88kSa/sec, one more click and I'm back on 0.14Sa/sec.

davidgraeff, 2018-03-09:
That's a correct warning, my compiler is completely silent about that one. And yes I can imagine that a one year old library can be a substantial obstacle on Debian, something that only time can fix.

davidgraeff, 2018-03-09:
The sispinbox widget that is responsible for the timebase/samplerate display is a maintenance hell. It crashed for me on the 6022 in various conditions, I expect this to not happen anymore. But its behaviour might be unexpected for other Dso samplerate/recordtime steps, like it is for yours. I will add a debug log message with the supported ranges, so that I can test with the exact same values.

phmarek, 2018-03-09:
I can't get any display.

And, in case that matters, the test signal on the 2250 is 2Vpp - the help message talks about a 1V signal, not sure whether that's amplitude or pp.

phmarek, 2018-03-09:
Are the ranges queried upon startup? My log box is always empty, so I'm not sure whether I'd even see startup-only messages there.

phmarek, 2018-03-09:
Is there an #openhantek channel on IRC so or, BTW?

phmarek, 2018-03-17:
24714ccffa04076b5d5b33dadacbfd7d0b936a43 still doesn't show any data, and has broken Samplerate/Timebase controls.

But the debug log now shows some entries -- though I didn't see how to export them, or how to get more details (sent data/received data).

Is there anything I can do to help? An USB trace, perhaps?

phmarek, 2018-04-08:
Any update on the samplerate issues? So many fixed on the OH2 branch, but sadly I can't even start to test.