OpenHantek/openhantek #157: Wish: measuring voltages

phmarek, 2018-01-13:
65f1202bf20910e9ba83fdf219e9b5aec715ec11 is great; now, if the Y position would be used too, it should be easy to mark two places via mouse clicks, to measure time resp. frequency *and* voltage differential at the same time!

Thank you for this software!

Hoagie572, 2018-01-13:
I like this idea too, but there”s a way to kind of do it already. Move the trigger slider up and down to see what voltage any point represents. Of course you lose trigger if you go too far so maybe pause the display first.

dendvz, 2018-01-13:
Well, the right way of enabling cursor measurements is to reimplement an idea from legacy Hantek SW: provide choice of 3 cursor shapes ('|', '-', '+') and attach markers to specific channel. Legacy SW has only one global pair of markers, we can go beyond that limitation and implement two markers per channel. Does not look too hard to implement, UX would be the most complicated part of this effort :)

dendvz, 2018-01-19:
Already started prototyping. Expecting first results for preview in a week or so.

- cursor selection: initially __on / off__ only, where __on__ is a rectangle; later on will add '+', '|' and '-' shape selector
- measurements grid is displayed as vertical 2x14 bar near left or right edge (position selectable via settings) of the scope view, e.g.

Markers | .
----- | -----
40 ms | .
__CH1__ | __ON__
20.00 ms | 2.0 V
__CH2__ | __OFF__
. | .
__MATH__ | __ON__
15.00 ms | 150 mV
__SP1__ | __ON__
20.00 kHz | 11 dB
__SP2__ | __OFF__
. | .
__SPM__ | __ON__
10.00 kHz | 13 dB

Click on cursor / marker name in a measurements grid activates the cursor for mouse events.

davidgraeff, 2018-01-20:
Omg, we will get some merge conflicts, whoever finishes his work first :D I'm working on GlScope at the moment as well ^^.

dendvz, 2018-01-20:
NP, the GlScope changes in this proto only affect GlScope::drawMarkers() :)

davidgraeff, 2018-01-20:
I have created a branch with my changes and will postpone them until yours are ready. The plan is to cut out grid/marker code into a separate class and draw to an offscreen texture, so that it can be reused by the exporter.

phmarek, 2018-01-21:
Also, how about having more keybindings available?
Eg. "m1" sets the first marker to the mouse position, "m2" the second one, "z1" moves the zero level of CH1 to the chosen vertical position, and so on.

This would be much better than the current "find-the-UI-with-the-mouse-and-fiddle-around".

Also, please bring "Ctrl-Q" back - this used to quit OpenHantek, but does so no more.

dendvz, 2018-01-21:
Keybindings make sense. At the moment I have control part (DsoWidget + scopesettings.h) more or less ready for preview. Not yet in: save / restore settings to / from ini file, cursor panel placement, dedicated toolbar button to show / hide all cursors, switching between cursor shapes. The latter is the first candidate for automation with shortcuts. Will also think of binding popup menu to right button click.

An immediate goal, however, is to merge my proto with branch prepared by David.

Screenshot (fake cursors, control part only):
[image]

dendvz, 2018-01-28:
PR #177