OpenHantek/openhantek #163: Crash on exit (OSX)

DanielO, 2018-01-14:


davidgraeff, 2018-01-14:
It crashes on Linux as well, but not all the time and therefore hard to debug

DanielO, 2018-01-14:
In this case I see..
(lldb)
frame #4: 0x00000001000b83e5 OpenHantek`USBDevice::disconnectFromDevice(this=0x00000001036bbd20) at usbdevice.cpp:101
   98  	    if (!this->handle) return;
   99
   100 	    // Release claimed interface
-> 101 	    libusb_release_interface(this->handle, this->interface);
   102 	    this->interface = -1;
   103
   104 	    // Close device handle
(lldb) print this
(USBDevice *) $0 = 0x00000001036bbd20
(lldb) print this->handle
(libusb_device_handle *) $1 = 0x0000000103919e00
(lldb) print this->interface
(int) $2 = 0

PR #164 (2018-01-15): Enforce constness/read-only access of specification through the code base

* Fixes a double free memory corruption. Fixes #163
* All DSO sub-models (e.g. DSO-5000, DSO-5000A) directly inherit from DsoModel.
* Remove DataArray util class. std::vector will perfectly do it and has the same
  memory allocation with common lib-std implementations (gcc, clang, VS).
* Make all short forwarding methods of USBDevice inline.
* Move BulkCommand+ControlCommand to hantekprotocol.
* Move usbprinterror method to USBDevice file.
commit 79301088 (2018-01-15): Enforce constness/read-only access of specification through the code base.

* Fixes a double free memory corruption. Fixes #163
* All DSO sub-models (e.g. DSO-5000, DSO-5000A) directly inherit from DsoModel.
* Remove DataArray util class. std::vector will perfectly do it and has the same
  memory allocation with common lib-std implementations (gcc, clang, VS).
* Make all short forwarding methods of USBDevice inline.
* Move BulkCommand+ControlCommand to hantekprotocol.
* Move usbprinterror method to USBDevice file.
closed 2018-01-15 by davidgraeff