OpenHantek/openhantek #165: GUI unusable - too slow to respond

tmshlvck, 2018-01-14:
GUI in the the last version (ab85a488ac1ee1a73d46d89779a71017a830ca9e) seems to be almost frozen right after start - reacts occasionally to certain mouse inputs, but it is generally unusable. Compiled on Ubuntu 17.10, using with DSO-2090.

Please see attached screenshot (it takes random time, usually over minute to show anything). I am also attaching my dpkg -l output.

[image]
[dpkg.txt](https://github.com/OpenHantek/openhantek/files/1629937/dpkg.txt)

davidgraeff, 2018-01-15:
Other OpenGL programs run fine on that system? But I agree that a failsafe mechanism should prevent freezing the GUI and show a message instead.

Leonleon33, 2018-01-15:
Maybe related ?
OpenHantek was working with version 9.01.2018_dc4cbc9 
but not with latest version.
Error opengl:
QOpenGLShader::compile(Vertex): 0:2(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES
*** Problematic Vertex shader source code ***
          #version 150
#define lowp
#define mediump
#define highp
#line 2
          in highp vec3 vertex;
          uniform mat4 matrix;
          void main()
          {
              gl_Position = matrix * vec4(vertex, 1.0);
              gl_PointSize = 1.0;
          } 
***
Ubuntu 17.10 on Dell Vostro 3700
$ glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Ironlake Mobile 
OpenGL version string: 2.1 Mesa 17.2.4
OpenGL shading language version string: 1.20
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 17.2.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

Is it possible to correct or do I have to stick to previous version ?

davidgraeff, 2018-01-15:
That's an unfortunate situation. Because it is a desktop system, it is compiled with OpenGL instead of OpenGL es on your system. But your hardware or software is not capable of OpenGL 3.3 with at least shader version 150. You need to force Qt to pick up OpenGL es during compilation.

Hoagie572, 2018-01-15:
Take a quick look at my comments on the fixgl pull request. If you make
those tweaks, you should be able to get it going with gles. On your system.

Geoff.

Leonleon33, 2018-01-15:
Thank you for your replies.
Hoagie572 is that the comment you suggest me to read?
"When compiling shaders, in glscope.cpp, I set the coreShaders boolean based on the QT_OPENGL_ES_2 definition and #defined QT_OPENGL_ES_2 at the top of the file."
As I am not a programmer, I would need a bit more specific howto.
I suppose I have to change line 176 in glscope.cpp ... 
Could you please tell me how, as well as the syntax to define QT_OPENGL_ES_2 at the top of the file.

Hoagie572, 2018-01-15:
Yep, that's the one.

Hoagie572, 2018-01-15:
Sorry, hit send before finishing the message on my phone. This is a very
dirty hack to see if it works.

put this line at the top of the file just before any #include lines
#define QT_OPENGL_ES_2

Then where it says bool coreShaders = lots of stuff;
change it to bool coreShaders = false;

Make sure it's lower case and has the semi-colon on the end.

That should get you running with OpenGL ES. It works on a Raspberry Pi 3
that way.

Regards,
Geoff.

tmshlvck, 2018-01-15:
Same compilation error for me.

Remaining questions: Other OpenGL apps (glxgears) works for me.

```
brill@slon:~/openhantek/build$ glxinfo | grep GLES
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
brill@slon:~/openhantek/build$ glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Ivybridge Desktop 
OpenGL core profile version string: 4.2 (Core Profile) Mesa 17.2.2
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 17.2.2
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 17.2.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
```

Hoagie572, 2018-01-15:
Sorry, but it's now way beyond my OpenGL knowledge. I just made a dirty
hack to run on the Pi and thought the error you gave looked like a
promising candidate for the same thing because it said it supported OpenGL
ES 1.0. Hopefully somebody who knows what they're doing will be able to
give proper guidance.

Geoff.

davidgraeff, 2018-01-15:
glxgears is a bad example, something more demanding and that needs to be from the last two years. I'll explain the situation:

OpenHantek before your mentioned commit used OpenGL fixed function operations. It was basically complained to OpenGL 1.2. This was good in a way that almost all systems with a graphics hardware were supported. But: The fixed functions were deprecated since OpenGL 3.0 (current OpenGL is 4.5) and in the attempt to get rid of them OpenGL ES was introduced. Some systems like the raspberry pi or all mobile systems only support OpenGL ES.

The mentioned commit modernized the OpenGL implementation and we use shaders, small programs on the GPU, now instead of the fixed draw pipeline. This is future proof and we can support the Raspberry and Android devices now. I have decided to go with a shader version that require at least OpenGL 3.3.

@Leonleon33: Your Dell Vostro 3700 is a 2010 model (8 years old by now). glxinfo says it only supports OpenGL 1.2, which does not meet the minimum requirement of OpenHantek anymore. But glxinfo also says that OpenGL ES 2.0 is supported. Theoretically OpenHantek would run in this mode, but you need a Qt installation that is compiled for OpenGL ES instead of Desktop OpenGL. But because you are using standard software for a desktop computer, your Qt is not configured that way. So you are out of luck, if you don't want to compile Qt yourself. 

@tmshlvck Your system is a more recent one and it should work for you. I guess that Qt decided for some reason that it want to use a software renderer instead of the real graphics hardware. Qt can emulate shaders on the CPU. But also your systems graphic stack may be in software rendering mode (lvmpipe). That's the reason why I ask you to try another OpenGl 3.3+ application, to see if your rendering in stack itself is misconfigured.

davidgraeff, 2018-01-16:
I have implemented a command line switch to choose between desktop OpenGL and OpenGL ES. Start OpenHantek like this: "OpenHantek --useGLES" if the normal version does not work for you.

dendvz, 2018-01-16:
PR #167 enables GLES auto-detection. Works on my HP 8440p (2011), OpenGL config is very close to one posted by @Leonleon33

Leonleon33, 2018-01-17:
Perfect ! Runs smoothly with the 'switchgl' branch , keeping the standard Qt. install.
Thank you very much David.

closed 2018-01-19 by davidgraeff