OpenHantek/openhantek #168: Windows build issues

MikePerreman, 2018-01-17:
In the cmake.lists file, make sure you add "set(CMAKE_PREFIX_PATH "C:\\Qt\\5.10.0\\msvc2015_64")"
also, I have this error:
`CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib`

Here is the full error, it happens right after downloading fftw:
`
Found OpenGL: opengl32   
Download FFTW for mingw64
[download 0% complete]
[download 1% complete]
[download 2% complete]
.......
[download 100% complete]
Microsoft (R) Library Manager Version 14.00.24215.1
Copyright (C) Microsoft Corporation.  All rights reserved.

LINK : fatal error LNK1171: unable to load mspdb140.dll (error code: 193)

CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib
Call Stack (most recent call first):
  cmake/fftw_on_windows.cmake:41 (CheckExitCodeAndExitIfError)
  openhantek/CMakeLists.txt:47 (include)
`

davidgraeff, 2018-01-18:
The mentioned file should be right next to link.exe of your VS2015 installation. Please search for it on your harddisk and verify.
The problem here is, I guess, that you modify CMAKE_PREFIX_PATH. Don't do that. The Qt paths can be set on the gui separately.

MikePerreman, 2018-01-18:
how would I set the paths in the cmake gui? also lib.exe is accessable anywhere, if i win+r, cmd, and type "lib" I get the context for lib.

davidgraeff, 2018-01-19:
[image]

MikePerreman, 2018-01-19:
Ok, You'll have to frogive me, as I'm still new at this, but I am getting this error in QtCreator, now:
```

[100%] Linking CXX executable OpenHantek.exe
MT: command "mt /nologo /manifest CMakeFiles\OpenHantek.dir/intermediate.manifest /out:CMakeFiles\OpenHantek.dir/embed.manifest /notify_update" failed (exit code 0x41020001) with the following output:
Copy fftw3 dlls for OpenHantek
Copy libusb-1 dlls for OpenHantek
Copy Qt5 dlls for OpenHantek
Setting up environment for Qt usage...
Remember to call vcvarsall.bat to complete environment setup!
"C:\Users\mikep\Desktop\OpenHantek\build-src-Desktop_Qt_5_10_0_MSVC2015_64bit-Default\openhantek\Debug\OpenHantek.exe" does not exist.

NMAKE : fatal error U1077: 'C:\Qt\5.10.0\msvc2015_64\bin\windeployqt.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
14:03:32: The process "C:\Program Files\CMake\bin\cmake.exe" exited with code 2.
Error while building/deploying project OpenHantekProject (kit: Desktop Qt 5.10.0 MSVC2015 64bit)
The kit Desktop Qt 5.10.0 MSVC2015 64bit has configuration issues which might be the root cause for this problem.
When executing step "CMake Build"
14:03:32: Elapsed time: 02:14.
```

Again, Im still new to this, but I did run vsvarsall.bat in admin command prompt, to no avail.

davidgraeff, 2018-01-20:
At the moment in your setup, QtCreator is again using the visual studio compiler. This is normally fine, but here you hit a bug in cmake: https://gitlab.kitware.com/cmake/cmake/issues/17444.

What we do now is to switch the used compiler to the default one:
* Close the project in QtCreator.
* Start the Qt installer (if necessary) to also install the mingw compiler.
* QtCreator:
   * Go to Tools --> Options.
   * Then navigate to "Build & Run".
   * Go to the Tab called "Kits". 
   * Hopefully there is a Desktop kit for mingw listed, select it and click the button "Make default" on the right side.
* Open the project again via File --> Open File or Project: Select the CMakeLists.txt of OpenHantek.

mike9755, 2018-01-20:
I met this problem days ago and I was coufued for couple days ,the project owner David inspired me, I think MikePerreman met the same problem as me, actually, there is a fatal problem when you generate VS project successfully with  CMake in windows, the default fftw package contains mingw compatible  library, so you need compile the fftw package with VS lib.exe before open the project in VS,the fftw website introduce the detailed compile method.
In the end,Davidgraeff is a very very kind man, I noticed he has fixed the problem in his recent commit, so the fastest way to fix this problem is cloning the newest code to your computer^_^

MikePerreman, 2018-01-22:
hmmm, this is the output that i get from qt creator, here is the full output, im not convinced that im doing this correctly: 
```

Running Windows Runtime device detection.
No winrtrunner.exe found.
Running "C:\Program Files\CMake\bin\cmake.exe -E server "--pipe=\\.\pipe\{d20ab46e-75d3-4349-b915-00148d7cb43a}" --experimental" in C:\Users\mikep\AppData\Local\Temp\QtCreator-uoJCGr\qtc-cmake-ggiuwmOV.
Starting to parse CMake project, using: "-DCMAKE_CXX_COMPILER:STRING=C:/Qt/Tools/mingw530_32/bin/g++.exe", "-DCMAKE_C_COMPILER:STRING=C:/MinGW/bin/gcc.exe", "-DCMAKE_PREFIX_PATH:STRING=C:/Qt/5.10.0/mingw53_32", "-DQT_QMAKE_EXECUTABLE:STRING=C:/Qt/5.10.0/mingw53_32/bin/qmake.exe".
The C compiler identification is GNU 6.3.0
The CXX compiler identification is GNU 5.3.0
Check for working C compiler: C:/MinGW/bin/gcc.exe
Check for working C compiler: C:/MinGW/bin/gcc.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Found OpenGL: opengl32   
Download FFTW for mingw32
[download 0% complete]
[download 1% complete]
[download 2% complete]
[download 3% complete]
[download 4% complete]
[download 5% complete]
[download 6% complete]
[download 7% complete]
[download 8% complete]
[download 9% complete]
[download 10% complete]
[download 11% complete]
[download 12% complete]
[download 13% complete]
[download 14% complete]
[download 15% complete]
[download 16% complete]
[download 17% complete]
[download 18% complete]
[download 19% complete]
[download 20% complete]
[download 21% complete]
[download 22% complete]
[download 23% complete]
[download 24% complete]
[download 25% complete]
[download 26% complete]
[download 27% complete]
[download 28% complete]
[download 29% complete]
[download 30% complete]
[download 31% complete]
[download 32% complete]
[download 33% complete]
[download 34% complete]
[download 35% complete]
[download 36% complete]
[download 37% complete]
[download 38% complete]
[download 39% complete]
[download 40% complete]
[download 41% complete]
[download 42% complete]
[download 43% complete]
[download 44% complete]
[download 45% complete]
[download 46% complete]
[download 47% complete]
[download 48% complete]
[download 49% complete]
[download 50% complete]
[download 51% complete]
[download 52% complete]
[download 53% complete]
[download 54% complete]
[download 55% complete]
[download 56% complete]
[download 57% complete]
[download 58% complete]
[download 59% complete]
[download 60% complete]
[download 61% complete]
[download 62% complete]
[download 63% complete]
[download 64% complete]
[download 65% complete]
[download 66% complete]
[download 67% complete]
[download 68% complete]
[download 69% complete]
[download 70% complete]
[download 71% complete]
[download 72% complete]
[download 73% complete]
[download 74% complete]
[download 75% complete]
[download 76% complete]
[download 77% complete]
[download 78% complete]
[download 79% complete]
[download 80% complete]
[download 81% complete]
[download 82% complete]
[download 83% complete]
[download 84% complete]
[download 85% complete]
[download 86% complete]
[download 87% complete]
[download 88% complete]
[download 89% complete]
[download 90% complete]
[download 91% complete]
[download 92% complete]
[download 93% complete]
[download 94% complete]
[download 95% complete]
[download 96% complete]
[download 97% complete]
[download 98% complete]
[download 99% complete]
[download 100% complete]
CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib
Call Stack (most recent call first):
  cmake/fftw_on_windows.cmake:41 (CheckExitCodeAndExitIfError)
  openhantek/CMakeLists.txt:47 (include)

Configuring incomplete, errors occurred!
See also "C:/Users/mikep/AppData/Local/Temp/QtCreator-uoJCGr/qtc-cmake-ggiuwmOV/CMakeFiles/CMakeOutput.log".
CMake Project parsing failed.
```

MikePerreman, 2018-02-04:
Hey, its been 2 weeks, and I cant figure out what ive done wrong. Can I please have some help?

thielj, 2018-02-19:
@MikePerreman, were you able to build this - and if so, would you be able to make your binaries available, please?

Hollowman1701, 2018-03-06:
@thielj hello, do you still need the binaries?

JoarGjersund, 2018-03-07:
I get the exact same errors as @MikePerreman trying to compile project both with Qt Creator and Visual Studio 2017. Did anyone find out what's wrong (or what I'm doing wrong)?

Trying to compile with Visual Studio outputs this error:
LINK : fatal error LNK1171: unable to load mspdb140.dll (error code: 126)

QTCreator on the other hand gives me this output:
CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib.exe:
Call Stack (most recent call first):
  cmake/fftw_on_windows.cmake:48 (CheckExitCodeAndExitIfError)
  openhantek/CMakeLists.txt:46 (include)

JoarGjersund, 2018-03-08:
@Hollowman1701 Yes, but now I'm  having problems installing the driver...

"There is no driver selected for the device  information set or element"

Hollowman1701, 2018-03-08:
@JoarGjersund 
which oscilloscope do you have?

JoarGjersund, 2018-03-08:
Hantek6022BE.

Hollowman1701, 2018-03-08:
[hantek driver.zip](https://github.com/OpenHantek/openhantek/files/1792810/hantek.driver.zip)
try this!

Hollowman1701, 2018-03-08:
[libusb_device.zip](https://github.com/OpenHantek/openhantek/files/1792851/libusb_device.zip)
try this inf

JoarGjersund, 2018-03-08:
Now it doesnt find the driver.  When I check the details of the device is says it is VID_04B5&PID_6022 so unfortunately still not working..

Hollowman1701, 2018-03-08:
ok. i know what your problem is

Hollowman1701, 2018-03-08:
which windows version do you use?

JoarGjersund, 2018-03-08:
I'm running Windows 10 Version 1709

Hollowman1701, 2018-03-08:
[hantek_mod_driver.zip](https://github.com/OpenHantek/openhantek/files/1792935/hantek_mod_driver.zip)
this must work

JoarGjersund, 2018-03-08:
Thanks @Hollowman1701 ! You're the king! Care to give me a quick explanation what was the problem?

Hollowman1701, 2018-03-08:
if I found out for myself, I'll tell you. :-)
i think for the first driver somethink is missing in the system32 folder

Hollowman1701, 2018-03-08:
is your openhantek now working?
or is it closing after you hit the ok button, like mine does?

JoarGjersund, 2018-03-08:
Now the device show up in the welcome GUI, but as you said it closes when i hit the ok button. 

onecore\com\combase\inc\comcataloghelpers.hpp(67)\combase.dll!00007FFA6B52FE7A: (caller: 00007FFA6B52E2D5) ReturnHr(1) tid(2864) 80004002 No such interface supported
'OpenHantek.exe' (Win32): Loaded 'C:\Windows\System32\sxs.dll'. Cannot find or open the PDB file.
mincore\com\oleaut32\dispatch\ups.cpp(2125)\OLEAUT32.dll!00007FFA6BA54D32: (caller: 00007FFA6BA54E96) ReturnHr(1) tid(329c) 8002801D Library not registered.
mincore\com\oleaut32\dispatch\ups.cpp(2125)\OLEAUT32.dll!00007FFA6BA54D32: (caller: 00007FFA6BA54E96) ReturnHr(2) tid(329c) 8002801D Library not registered.
'OpenHantek.exe' (Win32): Loaded 'C:\Windows\System32\oleacc.dll'. Cannot find or open the PDB file.
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\libusbk.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\hid.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\devobj.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\msasn1.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\crypt32.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\wintrust.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\setupapi.dll'
The thread 0xae0 has exited with code 0 (0x0).
The thread 0x2864 has exited with code 0 (0x0).
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\d3d11.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\dcomp.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\rmclient.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\twinapi.appcore.dll'
'OpenHantek.exe' (Win32): Unloaded 'C:\Windows\System32\DataExchange.dll'
The thread 0x3088 has exited with code -1 (0xffffffff).
The thread 0x2940 has exited with code -1 (0xffffffff).
The thread 0xcb0 has exited with code -1 (0xffffffff).
The thread 0x31bc has exited with code -1 (0xffffffff).
The thread 0x3458 has exited with code -1 (0xffffffff).
The thread 0x30d0 has exited with code -1 (0xffffffff).
The thread 0x16bc has exited with code -1 (0xffffffff).
The thread 0x2fcc has exited with code -1 (0xffffffff).
The thread 0x2828 has exited with code -1 (0xffffffff).
The thread 0x1278 has exited with code -1 (0xffffffff).
The program '[4824] OpenHantek.exe' has exited with code -1 (0xffffffff).

Hollowman1701, 2018-03-08:
[Release.zip](https://github.com/OpenHantek/openhantek/files/1793108/Release.zip)
try this.

JoarGjersund, 2018-03-08:
That one is working. Although I kind of wanted to be able to compile it myself so I could play around with the code. But Thanks!

Hollowman1701, 2018-03-08:
this release is older.
the latest code doesn't work on windows

Hollowman1701, 2018-03-08:
trust me. 
I have been trying for 1 month.
with 5 different qt versions and visual studio 2010 2013 2015 and 2017
i also build a qt version for windows xp and every time openhantek crashes after the ok button

davidgraeff, 2018-03-09:
I'm convinced now to setup a windows virtual machine and try myself

thielj, 2018-03-09:
@Hollowman1701 the build on AppVeyor is completing, but the binaries don't work. I get as far as OpenHantek updating the 6022BE's firmware and when I hit OK, the app terminates. Probably the same issue you're seeing. 

Guys, if you have trouble installing the driver, unplug the scope. Open the device manager, enable display of hidden devices and uninstall all instances of the scope you're seeing. Do a restart and repeat if necessary. Then plug-in the device in and install the WinUSB driver using Zadig (http://zadig.akeo.ie/): pick the HantekDSO6022BE (or Unknown Device, USB ID 04B5 6022), select the WinUSB driver, click install and re-plug the scope. Zadig installs signed libusb drivers!

@davidgraeff, thanks for looking onto this now. I think it would make sense to recommend Zadig instead of the current libusb driver instructions!

Hollowman1701, 2018-03-10:
@thielj yes, exactly the same problem. firmware upload done. click ok. goodbye
@davidgraeff here is the error just bevor the app crashes.
EXCEPTION_DEBUG_INFO:
           dwFirstChance: 1
           ExceptionCode: C0000005 (EXCEPTION_ACCESS_VIOLATION)
          ExceptionFlags: 00000000
        ExceptionAddress: 000007FEEFF520E8 libusb-1.0.000007FEEFF520E8
        NumberParameters: 2
ExceptionInformation[00]: 0000000000000000 Read
ExceptionInformation[01]: FFFFFFFFFFFFFFFF Inaccessible Address
Erste Ausnahme[First chance exception] bei 000007FEEFF520E8 (C0000005, EXCEPTION_ACCESS_VIOLATION)!

Hollowman1701, 2018-03-10:
@davidgraeff if you wan't to have a look. this is a log file from building with QT5.6.3/MSVC2015_64.
build process always the same.
i tried with QT5.6.3, QT5.7.1, QT5.8.0 and QT5.10.0 in combination with MSVC2013, MSVC2015 and MSVC2017. 32-bit and 64-bit.
in all builds same problem. firmware upload works, then hit ok and crash.
[build_log.txt](https://github.com/OpenHantek/openhantek/files/1798916/build_log.txt)

MikePerreman, 2018-03-12:
I am now having the same issue that @JoarGjersund was having, his solution is not applicable to me, as Im using VS 15, there is no related directory to what he described. Here is my error:

```
Running "C:\Program Files\CMake\bin\cmake.exe -E server "--pipe=\\.\pipe\{d8b19cb4-9cd4-4f56-b31a-2f7727e268cc}" --experimental" in C:\Users\mikep\AppData\Local\Temp\QtCreator-PHoDua\qtc-cmake-PhprDOxS.
Starting to parse CMake project, using: "-DCMAKE_CXX_COMPILER:STRING=C:/Qt/Tools/mingw530_32/bin/g++.exe", "-DCMAKE_C_COMPILER:STRING=C:/MinGW/bin/gcc.exe", "-DCMAKE_PREFIX_PATH:STRING=C:/Qt/5.10.0/mingw53_32", "-DQT_QMAKE_EXECUTABLE:STRING=C:/Qt/5.10.0/mingw53_32/bin/qmake.exe".
The C compiler identification is GNU 6.3.0
The CXX compiler identification is GNU 5.3.0
Check for working C compiler: C:/MinGW/bin/gcc.exe
Check for working C compiler: C:/MinGW/bin/gcc.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Found OpenGL: opengl32   
Download/Extract FFTW for 32bit
CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib.exe:
Call Stack (most recent call first):
  cmake/fftw_on_windows.cmake:48 (CheckExitCodeAndExitIfError)
  openhantek/CMakeLists.txt:46 (include)

Configuring incomplete, errors occurred!
See also "C:/Users/mikep/AppData/Local/Temp/QtCreator-PHoDua/qtc-cmake-PhprDOxS/CMakeFiles/CMakeOutput.log".
CMake Project parsing failed.
```

Hollowman1701, 2018-03-13:
@MikePerreman did you even read what is currently going on here?
maybe not. ok
to solve your problem.
open a command prompt.
go to the main openhantek folder, where cmakelist.txt is.
and then copy this text

set configuration=Release
set QT5=C:\Qt\5.10.0\msvc2015_64
set "CMAKE_GENERATOR_NAME=Visual Studio 14 2015 Win64"
set Path=%QT5%\bin;%Path%
md build
cd build
call "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" x64
cmake.exe ..\ -G "%CMAKE_GENERATOR_NAME%" "-DCMAKE_PREFIX_PATH=%QT5%" -DCMAKE_BUILD_TYPE=%configuration%
cmake --build . --config Release --target all_build
pause

0x8BitStorm, 2018-05-10:
NOTE: The paths below are for Visual Studio 2017 Community edition

So I was able to fix this enough to get a windows build by editing (cmake/fftw_on_windows.cmake)

First I added some debugging messaging before the exec command, I wanted to output message(${_vs_bin_path})

It spit out the path:

 C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.14.26428\\bin\\Hostx86\\x64\\lib.exe

I thought Hostx86 was odd since I was running on an x64 build of windows and trying to build x64 software, so I hard-coded my exec command to be pointed to lib.exe in the Hostx64 directory:

execute_process(
    COMMAND "C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.14.26428\\bin\\Hostx64\\x64\\lib.exe" ${LIBEXE_64} /def:${CMAKE_BINARY_DIR}/fftw/libfftw3-3.def /out:${CMAKE_BINARY_DIR}/fftw/libfftw3-3.lib
    WORKING_DIRECTORY "${CMAKE_BINARY_DIR}/fftw"
    OUTPUT_VARIABLE OutVar
    ERROR_VARIABLE ErrVar
    RESULT_VARIABLE ExitCode)
CheckExitCodeAndExitIfError("lib.exe: ${OutVar} ${ErrVar}")

This gave me the libfftw3-3.lib for x64 (I don't know if I can distribute this .lib, my guess is I can, source is unmodified)
[libfftw3-3.zip](https://github.com/OpenHantek/openhantek/files/1990084/libfftw3-3.zip)

With the binary, after a failed c-make, you can paste the .lib file provided in the <Cmake Bin Directory You Specified>\fftw  folder and replace your fftw_on_windows.cmake with the one attached here:
[fftw_on_windows.cmake.txt](https://github.com/OpenHantek/openhantek/files/1990092/fftw_on_windows.cmake.txt)

Which simply removes the cmake command, that should get you building.  I can't say if this is a visual studio or cmake configuration issue, since the path comes from this command.

get_filename_component(_vs_bin_path "${CMAKE_LINKER}" DIRECTORY)

So you may need to work around the cmake error for x64 hosts on certain visual studio versions.  I am not that well versed in cmake to submit a fix, I just found that it's a path issue.  Hope this helps.

0x8BitStorm, 2018-05-10:
@Hollowman1701 the crash you describe happens in windows because the usbdevice gets closed, and the device objects is unref'd through libusb, then when the device setup dialog gets closed it tries to open it again to connect to the oscope, but the windows libusb implementation must not like working with an unref'd device pointer, so it errors.  I temporarily fixed it by commenting out the call to unref, but I am working on a fix for this.  This bug here has lead me to the issue, you can see the comment about usb devices towards the bottom: https://github.com/OpenHantek/openhantek/issues/178

Hollowman1701, 2018-05-10:
@0x8BitStorm  yeah. a fix would be nice.
but, can you tell me where i have to commenting out what? 
i need to try this. NOW! :-)

thielj, 2018-05-10:
@0x8BitStorm @Hollowman1701 - guys, if you have a fix, let me have it, please. The automated builds in my fork (https://github.com/thielj/openhantek/) are working and I can create binaries within the hour!

0x8BitStorm, 2018-05-11:
@thielj @Hollowman1701 I checked in the fix for the usbdevice crash when starting the app initially.  It is an object lifetime issue, I think *nix/osx has an underlying reference in libusb that isn't part of the windows implementation, letting this code work on those platforms and not windows.  I would have to investigate more to be sure, or at the very least get a linux machine set up to test this on *nix before merging it in here.  But if @thielj 's builds are not affected by the 64 bit libfftw he should be able to get everyone windows binaries 😄

0x8BitStorm, 2018-05-11:
@Hollowman1701 if you have a build working and crashing at this point I attached the usbdevice.cpp file here.  Just rename it to remove the .txt extension.

place it in your openhantek\src\usb folder.

[usbdevice.cpp.txt](https://github.com/OpenHantek/openhantek/files/1993729/usbdevice.cpp.txt)

Hollowman1701, 2018-05-15:
@0x8BitStorm it works!!! you are a genius.

Hollowman1701, 2018-05-25:
@davidgraeff hello, this is now a working openhantek build. maybe you can put it to downloads.
tested under Windows XP SP3 32-Bit (Software OpenGL).
for other Windows OS, delete the opengl32.dll from Openhantek folder.
[Release.zip](https://github.com/OpenHantek/openhantek/files/2040985/Release.zip)

Hollowman1701, 2018-05-28:
@davidgraeff what is PR?

davidgraeff, 2018-05-29:
Pull Request. A pull request bundles commits to fix a particular issue and a core concept of git and github for easily giving back improvements to the upstream.

0x8BitStorm, 2018-05-31:
I created a pull request against theilj's repo, but he never accepted it.  I can create one agaisnt yours if you wish.

0x8BitStorm, 2018-05-31:
It's not for this bug but rather a crash on startup with the USB

davidgraeff, 2018-05-31:
A PR would be nice, of course.

circuit-specialists, 2018-07-27:
I'm having troubles compiling on windows. I'm using QT Creator 5.11.1. What am I missing?

```
Running "C:\Program Files\CMake\bin\cmake.exe -E server "--pipe=\\.\pipe\{9d337508-488d-4fa6-bcd8-ad549491266e}" --experimental" in C:\Users\CircuitPro7\Downloads\build-openhantek-master-Desktop_Qt_5_11_1_MinGW_32bit-Default.
Starting to parse CMake project, using: "-DCMAKE_CXX_COMPILER:STRING=C:/Qt/Tools/mingw530_32/bin/g++.exe", "-DCMAKE_C_COMPILER:STRING=C:/Qt/Tools/mingw530_32/bin/gcc.exe", "-DCMAKE_PREFIX_PATH:STRING=C:/Qt/5.11.1/mingw53_32", "-DQT_QMAKE_EXECUTABLE:STRING=C:/Qt/5.11.1/mingw53_32/bin/qmake.exe".
The C compiler identification is GNU 5.3.0
The CXX compiler identification is GNU 5.3.0
Check for working C compiler: C:/Qt/Tools/mingw530_32/bin/gcc.exe
Check for working C compiler: C:/Qt/Tools/mingw530_32/bin/gcc.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe
Check for working CXX compiler: C:/Qt/Tools/mingw530_32/bin/g++.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Found OpenGL: opengl32   
Download/Extract FFTW for 32bit
Windows-10.0.17134
CMake Error at cmake/fftw_on_windows.cmake:7 (message):
  Failed: lib.exe:
Call Stack (most recent call first):
  cmake/fftw_on_windows.cmake:67 (CheckExitCodeAndExitIfError)
  openhantek/CMakeLists.txt:51 (include)

Configuring incomplete, errors occurred!
See also "C:/Users/CircuitPro7/Downloads/build-openhantek-master-Desktop_Qt_5_11_1_MinGW_32bit-Default/CMakeFiles/CMakeOutput.log".
CMake Project parsing failed.
```

[CMakeOutput.log](https://github.com/OpenHantek/openhantek/files/2237483/CMakeOutput.log)

circuit-specialists, 2018-08-02:
So I found that portion of code in fftw_on_windows.cmake. It is looking for the VC++ version.
`COMMAND "${_vs_bin_path}/lib.exe" ${LIBEXE_64} /def:${CMAKE_BINARY_DIR}/fftw/libfftw3-3.def /out:${CMAKE_BINARY_DIR}/fftw/libfftw3-3.lib`

davidgraeff, 2018-08-03:
Hm, a lot of people run into this issue. And lib.exe is actually not doing much but creating a symbol text file. As soon as I get my hands on a Windows machine I will create the file and just add it to the repository and remove the entire `lib.exe` step.

circuit-specialists, 2018-08-08:
I've played around with it more, but have yet to have a working model.

If you need windows, windows 10 is free and doesn't require a code for install. Just use an iso, and don't use a key on install. It'll say it's unactivated, but it'll still work just as normal.

circuit-specialists, 2018-08-08:
So I did get further based on 0x8BitStorm previous thread. I've rewritten the "fftw_on_windows.cmake" file to specify the correct parameters. I'm getting an error at nearly the end of the build process, but I haven't had time to check it yet.

[fftw_on_windows.zip](https://github.com/OpenHantek/openhantek/files/2268648/fftw_on_windows.zip)

circuit-specialists, 2018-08-09:
So, I'm not entirely sure how to perform a pull request, this will be my first one, but I've submitted my changes to you for review. It compiles just fine now on windows.

LameLefty, 2018-08-25:
Can someone please explain the entire toolchain/compilation process for Windows in a step-by-step process? I compiled the December 2017 release last night in 5 minutes using Ubuntu but in Windows it's all such a convoluted mess of installers, downloads, reconfiguring paths and editing CMakeLists.txt and/or fumbling with CMake-gui … Ugh. I grabbed the zip file posted by hollowman1701 back in March and it seems to work fine (Win10-64 bit) after deleting the 32-bit DLL referenced in his post. I'd really like to be able to build the software myself. Qt's Windows Maintenance Tool wants to target either mingw53 32-bit or any of the MS Visual Studio installations. With just mingw53 32 bit selected, after installation Qt Creator can't ever seem to find the C and C++ compiler binaries for the default Desktop toolkit. Do I have to install MS VisualStudio and then target that in Qt Creator?

LameLefty, 2018-09-01:
Okay, so in the last week I've figured out how to build the code under Win10 and have a working application. Yay. But for some reason, my trigger slope indicators aren't displaying properly - rather, both indicators appear as question marks. Is this a glitch in the Qt version I'm building against (5.11.1), some weirdness with Win10, or … ? 

[image]

I'm open to suggestions from anyone who might have a clue. Thanks!

LuchkaOnTheMoon, 2024-12-18:
@LameLefty could you please share to the community your findings about building under MS Windows?