OpenHantek/openhantek #171: Relation between timebase and sample rate non-obvious

wom-bat, 2018-01-19:
Changing the timebase affects the sample rate in non-obvious (and wrong!) ways.  Setting the timebase to 1us/div changes the sample rate to 100ks/s --- which means one sample per 10 divisions.

I'm using a DSO6022BE on Debian 'sid', and the current tip-of-git tree.

stevedee, 2018-01-19:
I've just installed OpenHantek for the first time, and I am also having problems with the timebase.

When I start (or restart) OpenHantek, I can increase the Samplerate from 100kS/s to 48MS/s, and the timebase gets reset to 10us. If I then set the Timebase to 4us the Samplerate drops to 100kS/s and cannot be set to any other value.

My system; Lubuntu 17.10 with Hantek 6022BE, running OpenHantek from main webpage, which I think is 16dc684.

dendvz, 2018-02-21:
Please use cb11dd7 which makes timebase-to-samplerate recalc more predictable. I would also suggest to avoid direct Samplerate selection on 6022BE; the SW trigger algorithm needs samples beyond display window boundaries for effective edge detection.

closed 2018-03-09 by davidgraeff