OpenHantek/openhantek #176: Calibration enhancement

Leonleon33, 2018-01-23:
Hello,
Would it be possible to implement a calibration (automatic or manual ) ?
My DSO6022BE show a few mv on both channels when they are grounded, but not at the same level (Origin is obviously hardware)
I am thinking of a new option in settings menu, to offset a channel (automaticaly or manualy) by the voltage masured when probe is grounded.

davidgraeff, 2018-01-23:
The 6022Bx have self calibration functions and OpenHantek is using them. I planned on making the entire DeviceSpecification struct of the current device available in the settings including the offsets, but didn't have time for that yet. 

Would be awesome if you could contribute the offset part at least.

Leonleon33, 2018-01-23:
I am OK to contribute.
But you have to know that I am only an old "amateur éclairé" with not that much "light" on C++
Where to begin ?

davidgraeff, 2018-03-09:
Self-calibration is a new feature in the development branch openhantek2

Leonleon33, 2018-03-09:
I tried to compile openhantek2... it fails linking CXX:
----------------
[100%] Linking CXX executable OpenHantek
CMakeFiles/OpenHantek.dir/src/hantekdso/dsocontrol.cpp.o : Dans la fonction « int qRegisterNormalizedMetaType<Hantek::BulkCode>(QByteArray const&, Hantek::BulkCode*, QtPrivate::MetaTypeDefinedHelper<Hantek::BulkCode, QMetaTypeId2<Hantek::BulkCode>::Defined&&(!QMetaTypeId2<Hantek::BulkCode>::IsBuiltIn)>::DefinedType) » :
/usr/include/x86_64-linux-gnu/qt5/QtCore/qmetatype.h:1682 : référence indéfinie vers « Hantek::staticMetaObject »
......
followed by a lot of similar "undefined references to"
« Hantek::staticMetaObject »
« PostProcessing::staticMetaObject » 
« Dso::staticMetaObject »
------------------

I cannot figure out what  I am missing in my configuration.

Leonleon33, 2018-03-09:
I did not write intentionally in bold ... Apologies

davidgraeff, 2018-03-09:
Use the code tags next time :) can you please tell which compiler you are using?

Leonleon33, 2018-03-09:
Output of cmake:
-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found libfftw3:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/x86_64-linux-gnu/libfftw3.so
-- Configuring done
-- Generating done

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

Leonleon33, 2018-03-13:
Latest commit 24714cc compiles OK.