OpenHantek/openhantek #178: No windows binaries in 2017-12 release

pniedojadlo, 2018-01-31:
Hi there,

Unlike advertised in the readme, there are no binaries in the 2017-12 release. Any chance this will be fixed? I already spent many hours trying to fix some obscur configurations in cmake/libwinpthread-1 and cannot afford spending so much time on crappy windows softwares cohesion.

Anyway, thank you for the awsome work, everything built nicely on debian using 3 commands from build instructions (apt install && cmake && make). So bad I own a surface tablette I want to use with my hantek...

Pierre

davidgraeff, 2018-01-31:
The continuous integration service should create a binary for every release and upload it to the github page. Something went wrong on this end. But I have no time to look into this.

thielj, 2018-02-11:
Any other download location for this?

davidgraeff, 2018-02-19:
Because OpenHantek is a community project and there is no one volunteering to package this software, I'm afraid, we don't have binaries at the moment. The software can be build by yourself though.

closed 2018-02-19 by davidgraeff

thielj, 2018-02-20:
I've found this at the end of the build log [here](https://ci.appveyor.com/project/davidgraeff/openhantek):

```
"GitHub" deployment has been skipped as environment variable has not matched ("appveyor_repo_tag" is "false", should be "true")
Build success
```

Could this be the reason for the missing windows binary?

thielj, 2018-02-20:
Here is as far as I got:

(1) I forked the project and created a [Windows x64 binary](https://github.com/thielj/openhantek/releases), basically by zipping up the build\openhantek\Release folder. There are some issues with your appveyor.yml, mainly around identifying artifacts and including them into the deployment (see mine). Notifications also need a valid email address and you should also delete the API token in yours, recreate it and encrypt it. Appveyor provides a form for that. I also found the generated installer package to be broken, so I didn't use that.

(2) Installed WinUSB driver using [Zadig](http://zadig.akeo.ie/), i.e. downloaded the latest build, plugged in scope, ran Zadig and picked the HantekDSO6022BE (or Unknown Device, USB ID 04B5 6022).Then I selected the WinUSB driver, clicked install and re-plugged the scope.

(3) openhantek.exe recognizes and seems to communicate with the scope:

```
stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 5092 bytes, 68 segments, avg 74
reset CPU
libpng warning: iCCP: known incorrect sRGB profile
```

(4) When I OK to connect, the openhantek app closes.

davidgraeff, 2018-02-20:
Thanks for investigating. Zadig is a good option, yes. Can you please build a debug variant. The release version will just crash, the debug one will output some more information hopefully. Ideally you start OpenHantek from within qtcreator or visual studio and debug from there to see where exactly it fails.

thielj, 2018-02-22:
Both the release and debug builds terminate with 0 exit code without any additional (debug) output. The builds were done in Appveyor and I don't have the necessary tools installed locally to debug. If you could have a look, please: https://github.com/thielj/openhantek/releases

chertov, 2018-02-27:
I think in this line when we are trying to connect to the device we did call libusb_open twice. https://github.com/OpenHantek/openhantek/blob/57f55429a9dd072757ffc87d223cb0c0cbec6362/openhantek/src/main.cpp#L131

i get exception here:
https://github.com/OpenHantek/openhantek/blob/57f55429a9dd072757ffc87d223cb0c0cbec6362/openhantek/src/usb/usbdevice.cpp#L65