OpenHantek/openhantek #182: Segmentation fault on Ubuntu 16.04 amd64

DmitryPan, 2018-02-06:
Hello and thanks for great app!

I am trying to build and run OpenHantek on Ubuntu 16.04. All dependencies are installed as described in build readme. The cmake and make complete without any errors but running OpenHantek executable returns segmentation fault.
```
# ./OpenHantek 
Segmentation fault (core dumped)
```

I have also tried debugging it with gdb but these details are not very useful for me:
```
# gdb -q ./OpenHantek 
Reading symbols from ./OpenHantek...done.
(gdb) run
Starting program: /root/Downloads/openhantek/build/openhantek/OpenHantek 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6432580 in std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
```

The old build by Oliver Haag for Ubuntu Maverick runs without any errors but does not detect my 6022BL.

Does anyone succeeded in running any newer version of OpenHantek on Ubuntu 16.04 amd64 or any other Linux distro? Maybe someone can share working executable or DEB-package?

Thanks,
DP

DmitryPan, 2018-02-06:
Update: I have repeated all steps on Ubuntu 17.04 and 17.10 both amd64 arch and running as VMs using VirtualBox (16.04 is host OS). There are also no errors during compilation process. If I run OpenHantek with disconnected device, I can see "Select device" window. As soon as I connect my 6022BL (or run software with device connected) the segmentation fault happens. This time gdb gives more details.
```
gdb -q openhantek/OpenHantek
Reading symbols from openhantek/OpenHantek...done.
(gdb) run
Starting program: /home/d/openhantek/build/openhantek/OpenHantek 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffea3fe700 (LWP 11952)]
[New Thread 0x7fffe2d93700 (LWP 11953)]
[New Thread 0x7fffe2592700 (LWP 11957)]
[New Thread 0x7fffd3fff700 (LWP 11958)]
[New Thread 0x7fffd1f97700 (LWP 11961)]

Thread 1 "OpenHantek" received signal SIGSEGV, Segmentation fault.
USBDevice::connectDevice (this=0x555555cb2e80, errorMessage=...)
    at /home/d/openhantek/openhantek/src/usb/usbdevice.cpp:78
78	        if (interface->num_altsetting < 1) continue;
(gdb) bt
#0  USBDevice::connectDevice (this=0x555555cb2e80, errorMessage=...)
    at /home/d/openhantek/openhantek/src/usb/usbdevice.cpp:78
#1  0x00005555555b0da7 in DevicesListModel::updateDeviceList (
    this=0x555555c973e0)
    at /home/d/openhantek/openhantek/src/selectdevice/deviceslistmodel.cpp:74
#2  0x00005555555b4eec in SelectSupportedDevice::<lambda()>::operator() (
    __closure=0x555555cc6100)
    at /home/d/openhantek/openhantek/src/selectdevice/selectsupporteddevice.cpp:85
#3  QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, SelectSupportedDevice::showSelectDeviceModal(libusb_context*)::<lambda()> >::call
    (arg=<optimized out>, f=...)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:135
#4  QtPrivate::Functor<SelectSupportedDevice::showSelectDeviceModal(libusb_context*)::<lambda()>, 0>::call<QtPrivate::List<>, void> (arg=<optimized out>, 
    f=...) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:192
#5  QtPrivate::QFunctorSlotObject<SelectSupportedDevice::showSelectDeviceModal(libusb_context*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (which=<optimized out>, 
    this_=0x555555cc60f0, r=<optimized out>, a=<optimized out>, 
    ret=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h:169
#6  0x00007ffff61a1afe in QMetaObject::activate(QObject*, int, int, void**) ()
```
We can see that segmentation fault is caused USB connection. This problem may be related to virtualized OS and USB device redirection.

Any suggestions how to run OpenHantek on Ubuntu 16.04?

Thanks,
DP

DmitryPan, 2018-02-07:
Update 2: I have tried to compile and run different forks of this project. Some older ones run successfully but after connectiong device segmentation fault happened exactly as on 17.04 and 17.10 on VMs, so it's not VM or USB redirection related. I have also tried gcc/g++ version 5.4.0, 5.4.1, and 7.2.0 with no luck. It's crazy.

DmitryPan, 2018-02-07:
Thanks for the fast reply!
Yes, I have installed udev rules file. I'm running OpenHantek as root so there should be no problems with accessing USB. Few minutes ago I have found a solution. The problem was in
libusb-1.0-0-dev library. I installed both libusb-1.0-0 and libusb-1.0-0-dev from Ubuntu Bionic repo (ver 2:1.0.21-2) and no more segfaults

schlagenhauf, 2018-02-10:
I have the same issue (16.04, 6022BL).
The last commit where it does not crash is eaf29e5917db5c7edff9fe0b8dfc2.

DmitryPan, 2018-02-10:
As I told, I succeeded in compiling and running latest OpenHantek. There were 2 different problems:

1. Segmentation fault when running OpenHantek executable even with disconnected device.

phoyd, 2018-02-19:
Hi Dmitry,

See https://github.com/OpenHantek/openhantek/pull/186 for your SEGFAULT in 

std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6

davidgraeff, 2018-02-19:
The fix by @phoyd is merged. Please report back if it works for you now :)

closed 2018-02-19 by davidgraeff