OpenHantek/openhantek #184: CMake Error at openhantek/CMakeLists.txt:3

dr-apple, 2018-02-09:
Hi i comp. the code but i became erros:

cmake ../ -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt5/5.7.0 
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at openhantek/CMakeLists.txt:3 (find_package):
  By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/Users/drapple/Downloads/openhantek-2017-12/build/CMakeFiles/CMakeOutput.log".

What can i do ?

davidgraeff, 2018-02-19:
Someone with a mac system need to help here. Cmake didn't find Qt on your system. Are you sure it is at the given path? (/usr/local/Cellar/qt5/5.7.0)

rtoma, 2018-02-19:
Just did `brew install qt5` and it ended up in /usr/local/Cellar/qt/5.10.1. So it seems the /qt5/ path part was changed to /qt5/ after the readme was created.

To summarize: for me this worked: `cmake ../ -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.10.1`

davidgraeff, 2018-02-19:
If it doesn't work for you, please reopen this issue

closed 2018-02-19 by davidgraeff