OpenHantek/openhantek #188: Add a top-level error handler

gjedeer, 2018-02-23:
OpenHantek on my device (Ubuntu 16.04, Hantek 6022BL) crashes like this:

```
terminate called after throwing an instance of 'std::runtime_error*'
[1]    470 abort (core dumped)  OpenHantek
```
Only by running gdb I can tell that it had thrown an exception in openhantek/src/docks/HorizontalDock.cpp line 140. Maybe it would be helpful for users to add a top-level error handler displaying the exception message before crashing.

davidgraeff, 2018-03-04:
The console log should be used whenever a runtime_error is thrown. But the general idea is to just get rid of the error source. The runtime_error is just a symptom for an error some layers further down.

davidgraeff, 2018-03-09:
The specific runtime_error and the cause for it was removed in the openhantek2 branch.