OpenHantek/openhantek #195: When started from shell, program doesn't terminate on Debian when closing application

richardklingler, 2018-03-05:
The shell isn't released when OpenHantek is terminated on Debian 9.
You have to do a Ctrl-C to quit the application.

davidgraeff, 2018-03-09:
Can you please test towards this specific issue with the openhantek2 branch?

richardklingler, 2018-03-09:
cmake fails on Debian 9:

klingler@debian:/home2/Develop/openhantek/build$ cmake ../
-- Version: openhantek2/3490680
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:26 (find_package):
  Could not find a package configuration file provided by "Qt53DCore" with
  any of the following names:

    Qt53DCoreConfig.cmake
    qt53dcore-config.cmake

  Add the installation prefix of "Qt53DCore" to CMAKE_PREFIX_PATH or set
  "Qt53DCore_DIR" to a directory containing one of the above files.  If
  "Qt53DCore" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  openhantek/CMakeLists.txt:3 (find_package)

davidgraeff, 2018-03-09:
Yes, the dependencies have changed. You need to install the qt package that provides Qt3d.

richardklingler, 2018-03-09:
No Qt-5.8 yet on Debian 9 (o;

CMake Error at openhantek/CMakeLists.txt:13 (message):
  Minimum supported Qt5 version is 5.8.0!

davidgraeff, 2018-03-09:
True. It's debian. qt3D is the perfect match for OH to abstract away GPU shaders. Released with Qt 5.7 it only got stable with Qt 5.8 though :/

wom-bat, 2018-03-12:
For what it's worth, I can build and run OpenHantek on Debian sid without these issues.

uunicorn, 2018-07-22:
I have the same problem. Looks like a double-free issue.

closed 2018-07-24 by davidgraeff

ikolinahr, 2018-08-20:
My apologies but I am new here. I just tried building openhantek2 and encountered the same error about Qt53DCore discussed in this thread. I am running openhantek 'master' just fine on Ubuntu 18.04, and have installed the package "qml-module-qt3d" suggested above, but the cmake error persists. What am I missing?

PR #246 (2018-07-24): Reset the device pointer before libusb_exit().
unique_ptr deletes target pointer when variable scope ends.
For variable "device" this is the whole main() function.
So Device is destroyed after main() is finished.
Device descructor is calling libusb_close. According to documentation
no libusb function is allowed to be called after libusb_exit().

Fixes #195
commit ff3bb8f3 (2018-07-24): Reset the device pointer before libusb_exit().

unique_ptr deletes target pointer when variable scope ends.
For variable "device" this is the whole main() function.
So Device is destroyed after main() is finished.
Device descructor is calling libusb_close. According to documentation
no libusb function is allowed to be called after libusb_exit().

Fixes #195