OpenHantek/openhantek #196: macOS: "select device" dialog stays forever on first launch: "upload in progress"

richardklingler, 2018-03-05:
Launching OpenHantek on macOS 10.12.6 from shell shows:

stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 5092 bytes, 68 segments, avg 74
reset CPU

But dialog stays forever with the message "Upload in progress"...
Quitting the application and relaunching the dialog tells that the device is ready to use...

Un-/replugging the device causes it to fail again...

No crash after first launch, only when device is used...

davidgraeff, 2018-03-09:
macos seems to have issues with the libusb library, especially the libusb_release_interface call. Is it possible to update to a newer libusb version?

richardklingler, 2018-03-09:
brew on macOS has already installed latest version:

OSX86:~ me$ brew upgrade libusb
Error: libusb 1.0.21 already installed

TjardF, 2018-03-30:
if got exatly the same problem,
my specs: 10.13.1 & Hantek 6022be....

nosolobots, 2018-05-24:
Same problem on Kubuntu 16.04 and Hantek 6022BE... Any idea?

osmanypereira, 2018-05-24:
Please double check what the Hantek 6022BE is reporting as vendor ID. In my case the problem was that the Hantek changed the vendor id unexpectedly from "VID_04B5" to "VID_04B4". Don't ask me how, it does not make sense for me either, but correcting it in the libusb_device_inf and reinstalling the driver made it to load again. Weird!

hackermagnet, 2018-09-28:
> Please double check what the Hantek 6022BE is reporting as vendor ID. In my case the problem was that the Hantek changed the vendor id unexpectedly from "VID_04B5" to "VID_04B4". Don't ask me how, [...]

I got the same issue. Could you give a little more info on fixing it? Thank you

mwcarlis, 2019-01-10:
I appear to be having the same issue on Ubuntu Linux..
Linux mattc-Z97X-UD5H 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

ABonner, 2019-04-27:
For Mac users: In case it isn't clear from the initial description, at least for me, if you run OpenHantek one time, let the upload complete, then click "Close" button on the dialog, THEN run OpenHantek again without unplugging the device, it will work the second time.

fredericgermain, 2019-04-27:
Quick note on Linux : 
* Don't forget to install the /lib/udev/rules.d/60-hantek.rules file
* Maybe run "sudo udevadm control -R"
* Replug your device

For cmake projects, having a debug build is usually done with : "cmake -DCMAKE_BUILD_TYPE=Debug .."

oz-matt, 2019-08-04:
Still having this issue on Windows :[ Anyone know of a fix?

yurivict, 2019-11-20:
I have the same issue on FreeBSD.
What is it uploading?

Ho-Ro, 2019-11-20:
Let me explain in short what happens (at least with Hantek 6022 and compatible devices):
This device is based on the Cypress [EzUSB chip](https://www.cypress.com/products/ez-usb-fx2lp)
The scope doesn't have a permanent storage for application program, so the firmware has to be loaded every time after power-on, read more about the [EZ-USB FX1/FX2LP Boot Options](https://www.cypress.com/file/43391/download).

After 1st power-on the devive is seen on USB as VID/PID 0x04B4/0x6022. If openhantek detects this devive it starts uploading the firmware file (typical an intel hex file that is included in the resources of the Linux/Windows/FreeBSD/MacOSX binary, coming from [there](https://github.com/OpenHantek/openhantek/tree/master/openhantek/res/firmware)). After uploading the file a new USB enumeration takes place automatically - the device 04B4/6022 disconnects from bus and reconnects - now with VID/PID **0x04B5**/0x6022 (this VID/PID comes from the firmware). Until this point everything works fine.
Then openhantek searches for this new number and **as soon it is found** the *uploading...* message disappears. This is the normal behaviour.

**BUT** if there are some other devices on the same USB bus (the USB subsystem is structured in busses and ports as shown by `lsusb; lsusb -t`), this re-enumeration could be disturbed and the program doesn't find the device anymore (openhantek looks every second for the changed ID iirc, yes, I'm correct -> https://github.com/OpenHantek/openhantek/blob/master/openhantek/src/selectdevice/selectsupporteddevice.cpp). 
If you close the window **without** unplugging the scope and restart `openhantek` it will start a new search and subsequently it finds the scope with the correct VID/PID 04B**5**/6022.
[image]

See also https://github.com/OpenHantek/OpenHantek6022/issues/35
And now for something completely different: openhantek is unmaintained at the moment #277, but you could try my fork https://github.com/OpenHantek/OpenHantek6022.

HTH, Martin

yurivict, 2019-11-20:
How long does such upload normally take?

```vendor 0x04b4 product 0x6022``` never disconnects, it just stays this way.

ABonner, 2019-11-20:
For me at least, the upload takes about 2-3 seconds

yurivict, 2019-11-20:
For me it never ends, device never switches to any other ID.
I guess I'll need to debug this further.

Ho-Ro, 2019-11-20:
Have you also tried [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022), which recently [received support](https://github.com/OpenHantek/OpenHantek6022/pull/22) for FreeBSD?

yurivict, 2019-11-21:
No luck with OpenHantek6022 for me, it behaves the same way.

yurivict, 2019-11-21:
This line
```
int status = libusb_claim_interface(handle, 0);
```
fails but the error message isn't printed: ```libusb_claim_interface() failed: LIBUSB_ERROR_OTHER```

Ho-Ro, 2019-11-21:
Thx für reporting, I will check the code flow.

yurivict, 2019-11-21:
With the line ```libusb_set_auto_detach_kernel_driver(handle, 1);``` ```libusb_claim_interface(handle, 0);``` fails.

Removing the line ```libusb_set_auto_detach_kernel_driver(handle, 1);``` allows it to work.

I'll update the port with a patch for this.

Ho-Ro, 2019-11-21:
I moved the issue tracking into "my" domain, [#35](https://github.com/OpenHantek/OpenHantek6022/issues/35).