OpenHantek/openhantek #198: Any possibility of added compatibility for Hantek 6074 series?

SagittariusC, 2018-03-05:
I wasn't sure where to post this, sorry if its in the wrong spot.

I recently purchased the Hantek 6074BC, 70mhz, 1Gs/s. And I soon found out the software has limitations, namely vertical fine adjustments. It also has triggering instability....

Was just curious if this software will ever support this model.

Thanks

-SagiC

davidgraeff, 2018-03-09:
No new devices will be added, it will only be made easier for others to add a new model. This software is not commercial oriented and only people, actually owing a specific model can add it to the codebase and test it properly.

SagittariusC, 2018-03-09:
Ah, I'c, so it is possible to attempt to connected under our own power. How do I go about doing that?

I read about it a little in some other threads, but I have a feeling I will get nowhere without any significant coding experience... Is this true?

davidgraeff, 2018-03-09:
Unfortunately this is true at the moment. One of ongoing (projects)[https://github.com/OpenHantek/openhantek/projects] is to make it possible to add a device without coding experience. We are not yet there.

SagittariusC, 2018-03-10:
Good to know that it is on the plate at least, I look forward to it.

Thanks for your time

bianchifan, 2019-03-23:
Mayby this can help you:
[https://github.com/hackhantek/hantek6004](url)
Android APK for Hantek 6004 device family ;)