OpenHantek/openhantek #206: app window height fixed and not all visible on small screens

stojanos, 2018-03-15:
Is there a way to set app window size smaller. Window's width can be changed but height is fixed and therefore I can't see the full graph (tool buttons on the right can be tilled so no issue with that).
Is there a way to configure window's height in some config file?
I am runnnig OpenHantek on linux (Ubuntu 17.10)
Thanks.

davidgraeff, 2018-03-15:
Please make a screenshot :) the minimal height is determined by the docking windows, e.g the voltage channel dock

stojanos, 2018-03-15:
Interesting, if you dock every window suddenly you get option to resize window. Now I can see everything. I need to run app with root permission. Could this be reason why my settings are screwed with a first start.

stojanos, 2018-03-15:
I can't reproduce the bug any more :-(

davidgraeff, 2018-03-15:
Not related. The root user starts with default settings as well as all other users. I'll close this issue then

closed 2018-03-15 by davidgraeff