OpenHantek/openhantek #211: pre-built packages aren't actually built?

schweini, 2018-03-26:
I just downloaded the .tar.gz from the releases page, and the zip from the website, and both are pure source code, and no binaries seem in sight?

davidgraeff, 2018-03-26:
Some with spare time need to fix the build scripts, yup

closed 2018-03-26 by davidgraeff

schweini, 2018-03-30:
> @davidgraeff <https://github.com/davidgraeff> Why are you closing issues
> that are obviously not fixed, while at the same time advertising Windows
> support and binaries? If you don't want to support this platform, simply
> say so. We'll get it, we'll stop trying to help and we'll accept that you
> don't want any help on this.
>
> I had provided you with a working build script mid-February; however,
> there's a software bug preventing the binaries from running, as confirmed
> by other people.
>
> See the other closed or pending issues on this:
>
> #178 <https://github.com/OpenHantek/openhantek/issues/178>
> #168 <https://github.com/OpenHantek/openhantek/issues/168>

I would like to point out that I personally don't mind compiling, but
repeatedly linking to and prominently announcing pre-built binaries is
confusing.

davidgraeff, 2018-03-30:
All platforms have to compile not just Windows users. It is the most complicated platform to compile software on, that's true, but you have chosen to use Windows. 

The binaries page is only linked on the readme page. 

If someone wants to fix the continuous delivery scripts, please help us and provide a PR. The current buildsystem or another multi platform system need to be used though. Solutions for just one platform are not accepted, sorry. :)

thielj, 2018-03-30:
1. I'm a Linux user since '94, but for this occasion, I need a Windows app. Simple as that.
2. You've already received what you're asking for (https://github.com/OpenHantek/openhantek/issues/178#issuecomment-367145574), but here it comes again:
3. The AppVeyor continous delivery script to build the binaries is here: https://github.com/thielj/openhantek/blob/master/appveyor.yml
4. Proof of concept: the binaries are being uploaded here (32 and 64 bit, debug and release):
https://github.com/thielj/openhantek/releases
5. You need to adjust the AppVeyor script to include your credentials and you should use AppVeyor's website to encrypt them
6. You should revoke your current, unencrypted credentials as they can be used to upload malware to your GitHub releases.
7. There's a bug - in the source code - that makes the binaries useless. It's documented here https://github.com/OpenHantek/openhantek/issues/168#issuecomment-371988052 and maybe solved here https://github.com/OpenHantek/openhantek/issues/178#issuecomment-369004266
8. I also gave you a much improved solution to Windows driver issues (Zadig), see here: https://github.com/OpenHantek/openhantek/issues/168#issuecomment-371870929

And if there are any questions regarding any of the above, I'm happy to help with it.

davidgraeff, 2018-03-31:
Thanks, I will link this post to the readme. I do not have time to work on this right now though. That's why I'm asking for PRs. I'll add missing credentials of course. there's a robot github account with releases push access existing specially for this though.

schweini, 2018-04-09:
@thielj Are your binaries really useless, or do you have a functional Windows x64 binary at hand? I would like to try my 'scope on a windows machine.

thielj, 2018-04-09:
@schweini useless, at least on my system and confirmed by others.  not getting past the first dialog.

schweini, 2018-04-09:
@thielj your version from https://github.com/thielj/openhantek/releases does setecht the Hantek 6022BE in the initial drop down box, but says "Connection failed!".

The other release you linked to doesn't give me an error, but also doesn't show me the initial dialog. I just get an empty oscilloscope screen, and no matter how many buttons i press, I don't get a signal.

BasicScope and the Hantek software is showing the test signal.

This is on Windows 8 x64.

pyr0ball, 2018-12-06:
I just picked up one of these scopes and having the same issue as @schweini where the app will open, detects the DSO-6022BE, but says "connection failed"

When I try to install using the "try unsupported method" the software states the OS may not allow access, and that I'd have to install a "generic driver" first. Is that driver available somewhere?

bat52, 2019-04-12:
Same here with DSO-6022BL. I have "connection failed!" on W64 Release (tried many)