OpenHantek/openhantek #212: Build / cmake  error on Mint 18.03

schweini, 2018-03-26:
After installing the listed prerequisites on Mint 18.03, running "cmake ../" gives the following error:

CMake Error at CMakeLists.txt:50 (find_package):
  By not providing "Findlibusb.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "libusb", but
  CMake did not find one.

  Could not find a package configuration file provided by "libusb" with any
  of the following names:

    libusbConfig.cmake
    libusb-config.cmake

  Add the installation prefix of "libusb" to CMAKE_PREFIX_PATH or set
  "libusb_DIR" to a directory containing one of the above files.  If "libusb"
  provides a separate development package or SDK, be sure it has been
  installed.

Vascom, 2018-03-26:
Maybe you forgot install libusb-1.0-0-dev?

schweini, 2018-03-27:
Weird. No - I did install it, ajust like the instructions said. 

I rebooted, and now make is chugging along.

Thanks, anyhow!

closed 2018-03-27 by schweini