OpenHantek/openhantek #213: Hantek 6022BE offset error

borisbme, 2018-03-26:
Hi,

I have a Hantek 6022BE, I moved to linux, and channel 1 has a 100 mV (zero) offset (also present when probe is connected to GND) This shift is not present on windows using the official app. 

I'm using arch linux and compiled Openhantek from the source.

[https://snag.gy/QcIL7Y.jpg](https://snag.gy/QcIL7Y.jpg)

Vascom, 2018-04-05:
I am also have offset +45mV on both channels with 6022BL.

rmagomedov, 2018-07-27:
I also see +100mV offset on Ch1 of my 6022BE with probe connected to GND. In the windows application everything is fine. @borisbme @Vascom were you able to solve this problem? It looks like the official application does some calibration procedure on every startup or applies saved offset corrections?

rmagomedov, 2018-07-29:
I have found a simple workaround. I shifted vertical position of channel which has problem with offset and now the same sine wave is seen almost identically on both channels. When you hold Alt key, you can set this position precisely in order to visually compensate the offset error

rmagomedov, 2018-08-14:
Unfortunately, the workaround I described above is not quite good, because you need to adjust vertical offset every time you switch channel voltage ranges (10mV to 5V)

davidgraeff, 2018-08-16:
It is not abondoned. It is in maintenance mode as visible on the main readme page. This means no feature development. Only critical bug fixes and pull request reviews.

There are no paid developers on this project. Don't expect anything!

justlep, 2020-04-29:
I read about the self-calibration feature in #176, but looks like the topic is abandoned. 

Are there any plans to either have auto-calibration or allow for manually defining DC correction values for each channel (e.g. some millivolt integer in the config file or Windows registry)?

Is there/what is currently a recommended way of adjusting the DC error? (except for dragging the zero marker in the UI)

Ho-Ro, 2020-04-29:
Do you know #277? Please check [these hints](https://github.com/OpenHantek/OpenHantek6022/issues/55) about calibration of 6022 devices.

justlep, 2020-04-29:
I wasn't aware of those calibration hints - thank you for the link!
It's been some struggle for me with zero Python experience, but I got it working eventually.

Just in case someone else ever wants to calibrate his/her Hantek 6022BE on a Windows 10 machine, here is what I did:

1. install Python3
2. get the required `libusb-1.0.dll` file: 
   - download the v1.0.23 zip from
      https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.23/
   - extract `libusb-1.0.dll` from the zip file into `{Python install dir}/`
     (where `python.exe` sits)
4. prepare Hantek6022API:
    ```
     git clone https://github.com/Ho-Ro/Hantek6022API
     cd Hantek6022API
     pip install .
    ```
5. when I tried to run the calibration at this point, Python complained about missing firmware files.
    I could solve this by copying the whole `Hantek6022API\PyHT6022\HantekFirmware` folder to
   `{Python install dir}\Lib\site-packages\PyHT6022\HantekFirmware`
6. run the calibration:
    ```
   cd examples
   calibrate_6022.py -gec
   ```

I was then able to calibrate the scope providing the given voltages.
All works perfectly fine now and voltage readings are nicely accurate.

Again, many many thanks for your work! :-)

Ho-Ro, 2020-04-30:
I opened a new [issue #8](Ho-Ro/Hantek6022API#8) in the [right repo](https://github.com/Ho-Ro/Hantek6022API), to concentrate all discussion about the python tools under Windows.

justlep, 2020-05-04:
The issue link was broken, it is https://github.com/Ho-Ro/Hantek6022API/issues/8

I just commented there. Everything's working great now. Thank you.

leonmills, 2021-04-05:
I have the same issue. Not there on Window Hantek app and it's Channe 1 only. Can someone explain what the disparity is and why it just shows on OpenHantek? I ran the self calibration on the Windows version which gives no feedback thinking it might write some values to the EEPROM and then it would be fine on Ubuntu but no such joy.

Ho-Ro, 2021-04-06:
Are you talking about [openhantek](https://github.com/OpenHantek/openhantek) or [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022)? If you are using the latter, please report this problem to the  [right place](https://github.com/OpenHantek/OpenHantek6022/issues/new?assignees=&labels=&template=bug_report.md&title=) as this repo is not maintained, see #277

leonmills, 2021-04-06:
My apologies - you were right - wrong place.