OpenHantek/openhantek #217: Where are the releases?

skaftetryne, 2018-04-19:
I'm trying to locate some prebuilt binaries for Linux without success. The github pages tells me to look at https://github.com/OpenHantek/openhantek/releases, but there are no binaries there.

osmanypereira, 2018-04-24:
Hi; I'm also looking for compiled binaries for Windows, but could not find them. Anyone that could help?

skaftetryne, 2018-05-04:
The lack of prebuilt binary is stopping me from testing what looks like the best software for my scope :( Have tried to compile from sources by following the instructions but compilation fails.

osmanypereira, 2018-05-07:
Hi, I also tried to compile it using Qt, but since I never used this environment I could not make it to compile under Windows 10, and I do not have time to learn Qt, I then tried to compile with VS 2015, which I use everyday for my developments, but It didn't work either. 
The software that came with my Hantek is pretty basic and has bugs, so I will keep watching this project for availability of binaries and driver for Windows 10.
I appreciate the efforts of these developers in putting together what looks like a great software.

coozoo, 2018-05-24:
@osmanypereira 
Hi here found this thread, suppose it's better to write her )
Thanks for trying. 
So as you suggested I've added qt.conf inside (strange windows can't find libs before its nose :) ).
I've updated it here https://github.com/coozoo/openhantek/releases/download/Fixes/openhantek.7z
But you already have it so you don't need it :)
About VID yes it's strange device :) Long time ago I was wondered with that too when extracted firmware from this device ) 
Not sure how to get this on windows but it should be somehow close.

On linux it's works in such way.
When you inserting it will be identified as:
`ID 04b4:6022 Cypress Semiconductor Corp.`
After uploading firmware
`ID 04b5:6022 ROHM LSI Systems USA, LLC`

And till reconnect device it will be as 04b5

Theremingenieur, 2018-05-26:
I've also been looking for a readily built binary for macOS - obviously in vain which is highly disappointing. If the developers want their software to be successful, why don't they make the effort to upload stuff which is ready to use? Not everybody feels comfortable with compiling Qt and this application...

coozoo, 2018-05-28:
@Theremingenieur mmm

> If the developers want their software to be successful

Actually opensource software is not written to be successful, it's written because "oh I need this" or "oh ok ok I will be so kind to do this".

> why don't they make the effort to upload stuff which is ready to use

Because efforts it's time and you can't waste your time to build for every crap that exists in this world, macOS is not exclusion. If you bought mac and continue to use it then it's comfortable for you. If you stuck in simple building then it's not comfortable and you can consider to change it :) 

Maybe you don't know but the mac app can be compiled only on mac, you can't cross compile it. So when you using such a ... you should be ready that you won't find binaries  for it. 
I'm not mac hater they making perfect laptop cases in china :). 

So now about build:
1. option 
I have VB setup especially to build for mac
But I don't have possibility to check because it's remote PC (my 8 years old lap is going mad when emulating mac).
so you can try [OpenHantek_clang64.zip](https://github.com/coozoo/openhantek/releases/download/Fixes/OpenHantek_clang64.zip)
The only what I've checked that it's running, but how to use it under mac you need to determine by yourself. If you lucky it maybe it will work without dances :)

If you'll try it so please report if it works.

2. option
still you can build it by yourself
Default scripts are not preparing bundled app by default, so you will receive just a binary which should work fine only for you mac.
General steps:
- Install brew
- In brew you need to install (maybe few more):
 brew install fftw
 brew install libusb
 brew install qt5
 brew install cmake
- Then inside source folder
 $ mkdir build
 $ cd build
 $ cmake ../
 $ make -j2
Note: if you got that cmake not able to find qt then you need to configure CMAKE_PREFIX_PATH

davidgraeff, 2018-05-28:
Travis continuous integration builds the linux and mac version for us. But as I pointed out before, someone need to improve the script and make it a proper continuous delivery script that uploads artifacts to github.

Theremingenieur, 2018-05-28:
Thank you both for your replies! I downloaded OpenHantek_clang64.zip and it launches already. Will check if it communicates with my 6022BE when I'll be home after work. If not, I'll try option 2, but that's something I wanted to avoid. I can understand that Macs are not the principal target of your activity, it's a relatively small market. I like my 2011 MacBook Pro just because it is extremely robust and still runs with the newest macOS while most Win machines of that age would already be outdated.

Theremingenieur, 2018-05-28:
@coozoo : Your compiled version runs well on my MacBook Pro, thank you so much! I've also made the effort to compile the sources from this GitHub here, which works, too, but that doesn't render on my 1280x800 screen (too big) and no mouse and keyboard interaction is possible.

coozoo, 2018-05-28:
@Theremingenieur  
you are welcome
and thank you for trying :)
So you've tried to compile this master branch?
Mine works fine  1280x800 and one from this master doesn't on?
Is it close to this it's linux? [Jumping UI](https://github.com/OpenHantek/openhantek/issues/223)
Or it's simply doesn't respond to keypresses?

Theremingenieur, 2018-05-28:
The problem was that the version which I compiled from this master branch started with a ways too big window, no menu was there and it did neither react to mouse clicks nor to keyboard entries (wanted to close it with cmd+Q) and I had to kill it. Just launched that version again to make a screen shot and - oh miracle! - it works now. No idea what happened. Thus, I have now even two working versions, both with the same flaw: Whatever color I select for the grid, it is never visible. The axes are there which is ok, but a seeing a div grid would be nicer...

coozoo, 2018-05-29:
@skaftetryne 
OMG what's happened to this world linux user asked about compiled binary
you can try this [OpenHantek-i386.AppImage](https://github.com/coozoo/openhantek/releases/download/Fixes/OpenHantek-i386.AppImage)

if you on fedora it should work
if another distro then 50/50

coozoo, 2018-05-29:
@Theremingenieur
oh got my device to virtualbox emulator :)
Hm strange do you mean grids with barely visible dots?
because I see  them and color change works, but they're really small and hard to see them
Really don't you have this dots?
[image]

Theremingenieur, 2018-05-29:
Sorry, my bad. It's perhaps a language or comprehension problem. When I read "grid", I expected something like that:
[image]

davidgraeff, 2018-05-29:
@coozoo You need the Qt svg image extension to be installed. The phosphor icon is provided as an svg.

Theremingenieur, 2018-05-29:
@coozoo 
Yes, that was my drawing because I wanted to show you what I (naively) expected to see. I'll look for the dots tonight, though...
I saw the icon problem but I considered it as being a minor issue. The version which I compiled from this master branch shows it correctly. 
BTW: Both versions cause a "beautiful" sigterm crash when quitting with cmd+Q. Will take a screenshot tonight.

coozoo, 2018-05-29:
@Theremingenieur
Suppose for those almost invisible dots you can open issue, because it's really hard to see them on mac and no matter what color )

wanders78746, 2019-07-29:
@coozoo, @davidgraeff
THANKS for the Windows binaries. And thanks to all the OpenHantek developers - it looks amazing and I look forward to running it. But I continue to have driver issues on Windows 10. Is it just me?

Windows 10 continues to complain about unsigned drivers. I did the following:

1) Uninstalled all Hantek drivers from my system.
2) Edited the **libusb_device.inf** file per the embedded instructions.
3) Plugged in my DSO-2150 and tried to install the driver from the "winusb driver" subdirectory.

No joy.  I got the same kind of error message that I get when I try to install the OEM drivers:

[image]

Do I need to run this under Linux in a VBox? Or is there a solution to this Windows 10 signed driver hell of which I am unaware?

davidgraeff, 2019-07-30:
You can disable the signed driver requirement entirely in the boot manager menu but that is really not recommended. Instead the used libusb implementation would need to be swapped to something that uses the windows non kernel usb stack where no signed drivers are required. Unfortunately OpenHantek is unmaintained on nobody will that I guess.