OpenHantek/openhantek #220: Openhantek6022 Sample rate

Rader02, 2018-05-13:
In OpenHantek6022 *Sample rate* and *Time base* are entangled. If I change one then the other changes as well. So far I failed to record data with samling rates in range of seconds.

Is it a mistake in operation? 

Thanks for development!

domino-pl, 2018-06-02:
Yes. I have same problem(also 6022be) and it make this software USELESS for me. Please repair it because it looks wonderful but does not work!!!

davidgraeff, 2018-06-02:
Repair it yourself! It is utterly useless to request a "repair". You haven't paid anything for this software. It it at the same time very unfriendly how you speak to the developer community.

@Rader02: the 6022 protocol does not allow to specify sample rate and timebase separately. They are entangled on the protocol level. The only thing OpenHantek can do here, is to simulate that they are separate, but there is no code doing that for now.

domino-pl, 2018-06-02:
I'm really sorry I've hurt you. I didn't want to do that. Mayby it's my bad English. I wanted to pray for, no request (dictionary shows it's better word).

I've written in order to inform you that was't unique situation. In fact it is impossible to use time base bigger then 1 or 2 seconds (orginal software could do that but works only with MS Windows) and i think the developer would like to know about error like that which make program really useless in many situations. I belived it was rather literal error and something was in wrong place.

In spite of that I have large respect for developers like you. You created great, big project but unfortunately that one very important think dosn't work correctly.

dendvz, 2018-06-18:
Lowest sample rate supported by Hantek 6022Bx HW+FW is 100kS/s. It is not an issue of OpenHantek software, anyway.

closed 2018-06-18 by davidgraeff