OpenHantek/openhantek #221: error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES

coozoo, 2018-05-20:
Hello
First of all I want to thank you guys for software you've did great job :)

This error appeared for me when I've compiled it and tried to launchit on my laptop:

`
QOpenGLShader::compile(Vertex): 0:2(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES
`
On the next platform:

```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Version: 17.2.4
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL version string: 2.1 Mesa 17.2.4
OpenGL shading language version string: 1.20
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 17.2.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
```

So I've made some fix for that. (Forgive me if I'm doing something wrong it's my first attempt to intrude into another project :) )

And as I absolutely dumb in a OpenGL it's better to verify what I've did few times )

So here's main moments of fix:
1. In main function there is useGles hardcoded as false but the code that follows by this variable all the time returns true. And suppose you should run into this part only when it's true, 
- so I can guess that should be if there :)
- and commented useGles = p.isSet(useGlesOption); because it's the part that actually returns true in any case because we make it set and doesn't mater what value is inside false or true.
2. It's actually fix for OpenGL ES (as previous state everytime was true suppose OpenGL ES wasn't tested :)).
When I've tried to use it.
I've got another error from OpenGL that precision should be defined for vertex. I've tried to set it bu got another error and didn't found a way to handle it out.
So  the way how I did fix just by changing version from 100 to 120 where you don't need to set precision. 
I don't like such fix because that means decreasing of supported devices...
After that OpenGL ES works fine.
3. It's the fix for an error that initially I've got about GLSL 1.50 not supported sure it's not supported :) But suppose such kind of software should run anywhere it shouldn't use latest OpenGL at least by default.
So I've modified vertex/shaders code to version 120 (I'm not sure about correctness, but it's working :) ).
4. And the last one moment is smallest.
It's set version to minimum supported by QT for compatibility (at least I think it's a good).
`format.setVersion(format.majorVersion(), format.minorVersion());`
But the best solution it would be determine version of opengl before setting up useGles decide what of opengl version we should use and set version appropriately.

So after everything done additionally to previous platform I've verified this fix for another one (much more newer) with such configuration:
```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 390.48
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 390.48
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 390.48
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

I wanna try it for some andoird device once I will find how to compile it for device.
PS now need to find how to pull request and attach it somehow :)

davidgraeff, 2018-05-28:
I think the fix should look different. Your old platform supports: 
```
OpenGL shading language: 1.20
OpenGL ES shading language: OpenGL ES 1.0.16
```
Which means it supports Open GL Shaders up to version 1.2 (Openhantek needs 1.5 at least) and it supports Open GL ES Shaders up to 1.0 (that's fine for Openhantek).

Shader version to OpenGL version table is here: https://github.com/mattdesl/lwjgl-basics/wiki/glsl-versions#glsl-versions

OpenHantek will only run with OpenGL ES on your old platform. Use the "--useGLES" switch or ask Qt in c++ which shader version is supported and implement an automatic switch to OpenGL ES :)

Ho-Ro, 2019-02-12:
Hi, unfortunately mesa, the OpenGL implementation for linux is very conservative in what it reports as its version. Even the latest version in testing (18.3.2) reports on my Lenovo T400:
```
$ glxinfo | grep OpenGL
...
OpenGL version string: 2.1 Mesa 18.3.2
OpenGL shading language version string: 1.20
...
```
even if it is able to deliver 4.5 functionality.
But you have the possibility to tweak this behaviour with some environment value magic:
```
$ MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150 glxinfo | grep OpenGL
...
OpenGL version string: 3.2 (Compatibility Profile) Mesa 18.3.2
OpenGL shading language version string: 1.50
...
```
Calling OpenHantek with these env settings works well and the screen reports OpenGL rendering.
The other way round was already mentioned above -> --useGLES (or patch it in your git / github).

Martin

@coozoo : I found the shader change 150 -> 120 in your commit https://github.com/coozoo/openhantek/commit/01b42d3db2921a0aa83ea6808147ea6257de3f28 - I'll have a look when I have some spare time.

Ho-Ro, 2019-05-02:
The @coozoo solution is in the active repo and is used if the run-time check detects OpenGL version < 3.2 at start.
https://github.com/OpenHantek/OpenHantek6022/commit/727950b354a24d30071df086d1a9b32d39f1dcfd

test2git, 2019-07-13:
Hello,

This program looks great. I have downloaded the current master code and i compiled it, but it doesn't work for me. I have the same versions of OpenGL / OpenGL ES

$ glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Q35 
OpenGL version string: 2.1 Mesa 13.0.6
OpenGL shading language version string: 1.20
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 13.0.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

$ OpenHantek
compile shaders
QOpenGLShader::compile(Vertex): 0:2(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES
(...)

$ OpenHantek --useGLES

Then it runs, does not show errors and the oscilloscope initialization seems ok (the led gets green) but the window doesn't redraw. There is no signal and it only redraws when i resize it manually.

[image]

[image]

I'm programmer but i have no experience with linux user interfaces and Qt/OpenGL...

Thank you.

Ho-Ro, 2019-07-14:
Hi,

Your OpenGL version is quite old (Mesa 13.0.6), which system do you use?

    OpenGL version string: 2.1 Mesa 13.0.6
    OpenGL shading language version string: 1.20

You can check if my environment magic from above can help you:
`$ MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150 OpenHantek`

Otherwise please try the glscope.cpp part of this commit: https://github.com/OpenHantek/OpenHantek6022/commit/97540dfcb80318a218786c4679f20fabfe5ef1e8
If it works you should implement a run-time check for the OpenGL version and select the appropriate shader code: https://github.com/OpenHantek/OpenHantek6022/commit/727950b354a24d30071df086d1a9b32d39f1dcfd

This repo is unmaintained and I forked it half a year ago and did a lot of development since. Unfortunately my target is only Hantek 6022 and I removed all other devices from the code some time ago. But you can try the [last legacy branch](https://github.com/OpenHantek/OpenHantek6022/tree/last_legacy_support) of my fork, it's the snapshot before the "great cleaning".

HTH Martin