OpenHantek/openhantek #229: Bundle app for Mac

coozoo, 2018-06-05:
Hi,
It's modifications to cmake that allowed to get bundle app.
Mainly oriented  to using brew but actually if include fftw and libusb libs and if you have  your own qt and cmake it's possible to compile without brew :)

So first of all the easiest way you need brew:
``` bash
$ brew install cmake
$ brew install qt5
$ brew install fftw
$ brew install libusb
```
NOTE  If you already have them installed then make sure your brewed qt5 is updated some old versions has broken macdeployqt so you will get broken bundle

1. At the beginning of script we need  to add location to our QT environment (at least brewed cmake 3.11.2 unable to find) 

``` cmake
if(APPLE)
    list(APPEND CMAKE_CXX_FLAGS "-mmacosx-version-min=10.9 ${CMAKE_CXX_FLAGS}")
    # Default brew QT location
    set(QT5_LOCATION "/usr/local/opt/qt5" CACHE)
	if(EXISTS ${QT5_LOCATION})
	    list(APPEND CMAKE_PREFIX_PATH ${QT5_LOCATION})
	    set(MAC_DEPLOY_QT_TOOL "${QT5_LOCATION}/bin/macdeployqt")
	else()
	     message(FATAL_ERROR "Install brew and Qt5 by running:\n brew install qt5 \n(Optionally you can change cmake ../ -DQT5_LOCATION=<yourQt> variable to your own Qt version)")
	endif()
endif()
```
This will allow to modify QT5_LOCATION variable by running cmake with parameters like this example:
`
$ cmake ../ -DQT5_LOCATION="/Users/admin/Qt5.7.0/5.7/clang_64"
`

2. Add MACOSX_BUNDLE to add_executable command

3. Set parameters of bundle info.plist
``` cmake
if(APPLE)
    set(MACOSX_BUNDLE_BUNDLE_NAME ${PROJECT_NAME})
    set(MACOSX_BUNDLE_EXECUTABLE start.sh)
    set(MACOSX_BUNDLE_INFO_STRING "${PROJECT_NAME} Osciloscope Software")
    set(MACOSX_BUNDLE_LONG_VERSION_STRING "${PROJECT_NAME}")
    set(MACOSX_BUNDLE_SHORT_VERSION_STRING "${PROJECT_NAME}")
    set(MACOSX_BUNDLE_BUNDLE_VERSION "${PROJECT_NAME}")
    set(MACOSX_BUNDLE_ICON_FILE openhantek.icns)
endif()
```
4. Change install target for apple
- copy icon
- deploy all ddl using macdeployqt
- create starter script and make it executable
- fix info.plist to use starter script

``` cmake
if(NOT APPLE)
    install(TARGETS ${PROJECT_NAME} RUNTIME DESTINATION "bin")
elseif(APPLE)
    install(TARGETS ${PROJECT_NAME} BUNDLE DESTINATION .)
    if(EXISTS ${MAC_DEPLOY_QT_TOOL}
	add_custom_command(
	TARGET ${PROJECT_NAME} POST_BUILD
	COMMAND ${CMAKE_COMMAND} -E copy "${CMAKE_SOURCE_DIR}/openhantek/res/images/openhantek.icns" "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/Resources/openhantek.icns"
	COMMAND "${MAC_DEPLOY_QT_TOOL}" "${PROJECT_NAME}.app" "-verbose=3"
	COMMAND printf "'#!/bin/bash\\n'" > "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/MacOS/start.sh"
	COMMAND printf "cd '\"$$\{0%%/*}\"\\n'" >> "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/MacOS/start.sh"
	COMMAND printf "'./${PROJECT_NAME}\\n'" >> "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/MacOS/start.sh"
	COMMAND chmod 777 "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/MacOS/start.sh"
	COMMAND plutil -replace CFBundleExecutable -string start.sh "${CMAKE_BINARY_DIR}/openhantek/${PROJECT_NAME}.app/Contents/Info.plist"
	WORKING_DIRECTORY "${CMAKE_BINARY_DIR}/openhantek"
	COMMENT "Deploy and copy app icon to bundle app")
    else()
	message(STATUS "No ${MAC_DEPLOY_QT_TOOL} found. So no bundle.")
    endif()
endif()
```

That's all
after executing 
$ cmake ../
$ make -j2

we can find bundled app and launch it
$ cd openhantek
$ open OpenHantek.app

davidgraeff, 2018-06-06:
Hm, are you sure, that CPack cannot generate the macos bundle for us? Doing that manually can already fail in the next version of MacOS.

coozoo, 2018-06-06:
I thought that set exec type to MACOSX_BUNDLE  should be enough to generate bundle app

but you can see that's all what you get just base structure
[image]

Not sure, didn't find such options in cpack now I see. I will try later if I have some time. But from fast view I don't see functions that will deploy it
[CPackBundle](https://cmake.org/cmake/help/v3.2/module/CPackBundle.html)

But actually the only one tool that maybe mac able to broke in a future plutil. Or yes they can create some security that will broke it :)

coozoo, 2018-06-07:
Oh finally read it the doc is terribly unclear how to use it...
So I didn't figure out how to get it work.
But what I've understood from doc that using cpack looks more manually then method provided here with macdeployqt.

[cpack/PackageGenerators](https://gitlab.kitware.com/cmake/community/wikis/doc/cpack/PackageGenerators)
Just from doc(as i didn't find a way to make it work)
1. terrible thing 
You need to use hardcoded precreated Info.plist stored in resources
2. you can't use MACOSX_BUNDLE generated plist because
`Important Note: Do not use the MACOSX_BUNDLE property on
executables that will be packaged using the bundle-generator!
Specifying MACOSX_BUNDLE creates a separate bundle for each individual
executable at build-time; the structure of these bundles becomes
redundant when the bundle generator consolidates multiple executables
into a single bundle.`
3. And from doc even after that when you got bundle you need to run fixup_bundles to put all libs inside bundle which looks more manual then standart qt deploy way (and from google as I see ppls has problems with right lib connections..)

so i'm surrender on it