OpenHantek/openhantek #233: Screen flashing when running openhantek

paulnewall, 2018-06-09:
On my laptop hp presario CQ60 with nvidia graphics, running xubuntu 18.04, the display driver is noveau. I get flashing of the screen when I run openhantek. I don't have this problem with any other application.
The flashes look like the bottom part of the screen briefly becomes white. The height of this bottom part varies randomly from flash to flash. The flashes repeat around twice a second.
Any ideas what I should try to fix this?

paulnewall, 2018-06-10:
The behaviour seems to be the same with OpenGL es (starting from command line with OpenHantek --useGLES)

ksa1952-git, 2020-06-29:
> The behaviour seems to be the same with OpenGL es (starting from command line with OpenHantek --useGLES)
had this same behaviour on fedora 32

Ho-Ro, 2020-06-29:
-> #277