OpenHantek/openhantek #236: No data displayed

shingarov, 2018-06-26:
Trying to run openhantek from current master with 5200A.
Firmware is successfully uploaded and the program starts and the scope starts data acquisition (green led flashing); when changing voltage limits, relays click when they should.
However, the ui draws no lines whatsoever.
I inserted some debug print statements to see what code is even reached.  For example, I inserted a debug print at the beginning of retrieveChannelLevelData() and device->controlRead() is called successfully but control only gets to that point once at the beginning of execution.  I also inserted another print at the beginning of PostProcessing::input() and that function is never reached.
Any suggestions how to debug this further?

davidgraeff, 2018-06-26:
There were a lot of changes to make the 6022 work. And nobody had a 5200 to test the changes with.
Could you debug step after data acquisition to see where it gets lost?

shingarov, 2018-06-26:
You mean stepping in gdb?  Yeah I can do that although I think it's going to be painful because I am not familiar with the openhantek codebase.

I am thinking about maybe an alternate approach, to start with a commit which is known to work with 5200 and bisect.

Do you know of a commit when the 5200 worked?

davidgraeff, 2018-06-27:
Bisect is sometimes a good approach, yes. Not in this case though. There are a lot of style/format and file move code changes since the very original OpenHantek version.
I would not recommend a pure gdb stepping. Use QtCreator/VisualStudio for fast code navigation while stepping.

Start here: https://github.com/OpenHantek/openhantek/blob/master/openhantek/src/hantekdso/hantekdsocontrol.cpp#L175 in `HantekDsoControl::getSamples(unsigned &previousSampleCount)`

This should return back to the state machine in https://github.com/OpenHantek/openhantek/blob/master/openhantek/src/hantekdso/hantekdsocontrol.cpp#L1209:
```cpp
        case CAPTURE_READY5200: {
            std::vector<unsigned char> rawData = this->getSamples(expectedSampleCount);
            if (this->_samplingStarted) {
                convertRawDataToSamples(rawData);
                emit samplesAvailable(&result);
            }
        }
```

* See if `convertRawDataToSamples` does something with the raw data and not erroring out.

As soon as `samplesAvailable` is emitted, the data should arrive in the post processing. I do not expect a problem there, 'cause it works for other DSOs.

henrik1g, 2019-03-08:
Did you have any success? I have a similar problem with the DSO-2250. Some very old versions of openhantek work. But anything current fails with symptoms similar to yours.
(Firmware upload, green led, relays clicking, all there, but no data on the screen)

milahu, 2020-07-22:
> Use QtCreator/VisualStudio for fast code navigation while stepping.

thanks, will try that

> I have a similar problem with the DSO-2250.

same here

i only see the voltage curve for math (CH1+CH2) but no voltage curve for CH1
CH1 is connected to CAL (calibration signal), 1KHz square wave 2Vpp
CH2 is connected to GND

spectrum for CH1 and math is visible

[image]

i use version 45521b4 cos my GPU only supports GLSL 1.20
```
$ glxinfo | grep -i "shading language"
OpenGL shading language version string: 1.20
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
```

Ho-Ro, 2020-07-22:
Unfortunately this repo is not supported at the moment (#277), but you can try this hint from #221:

> You can check if my environment magic from above can help you:
`$ MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150 OpenHantek`

> Otherwise please try the glscope.cpp part of this commit: https://github.com/OpenHantek/OpenHantek6022/commit/97540dfcb80318a218786c4679f20fabfe5ef1e8
> If it works you should implement a run-time check for the OpenGL version and select the appropriate shader code: https://github.com/OpenHantek/OpenHantek6022/commit/727950b354a24d30071df086d1a9b32d39f1dcfd

milahu, 2020-07-22:
> MESA_GL_VERSION_OVERRIDE=3.2 MESA_GLSL_VERSION_OVERRIDE=150

yes! its working now (version e7e0c7b), i just have to wait 5 seconds for the GUI to load
tried it before, but gave up too quick
is it a good idea to automate this hack? (more aggressive detection of GLSL support)

version 97540df works without the env-variables
but only with the `--useGLES` flag, also 5 seconds for app start
without that flag it fails with ``QOpenGLShader::link: error: vertex shader lacks `main'``

Ho-Ro, 2020-07-22:
IIRC I started from [this patch](https://github.com/coozoo/openhantek/commit/01b42d3db2921a0aa83ea6808147ea6257de3f28) that enabled OpenGL SL versions < 1.50. I added the version detection later because there was a report about regressions with (much) higher OpenGL SL versions (>2.00 ?). But you could try for your system. 
My Debian stable on a ThinkPad T400 with Intel graphic reports 1.20 capability even if the installed OpenGL version supports at least 1.50.
I do not know much about OpenGL SL programming, it was a simple C&P from @coozoo.

Martin