OpenHantek/openhantek #237: DSO5202P on MacOS

WilixLead, 2018-07-04:
I follow all steps in https://github.com/OpenHantek/openhantek/blob/master/docs/build.md#apple-macosx  
But got this error:  
[image]

Device connected via usb

bikermikem, 2018-07-06:
I have a working build but you must have the usb drivers installed on your Mac which they would appear not to be on your system.  Admittedly it took me over a week to get this working but then I’m not a programer and the instructions, as usual, assume user is knowledgeable about such things.  Also info for building seems to be a bit spread out over the OpenHantek website.  I’m going to try to put together all the steps I did and post it somewhere useful.  But for the moment the binary I built, the terminal output for it, and a screenshot & a video of screen are at 

https://www.dropbox.com/sh/ide1nsg8yjmi2ra/AABkzzN9Wtwxu42td6KbxUDna?dl=0 <https://www.dropbox.com/sh/ide1nsg8yjmi2ra/AABkzzN9Wtwxu42td6KbxUDna?dl=0>

WilixLead, 2018-07-07:
Where get this driver? My DSO5202P listed on system like this:  
```
Gadget Serial v2.4@14100000  <class AppleUSBDevice, id 0x1000522e9, registered, matched, active, busy 0 (1 ms), retain 12>
        {
          "sessionID" = 557095611963799
          "iManufacturer" = 1
          "bNumConfigurations" = 1
          "idProduct" = 20570
          "bcdDevice" = 9264
          "Bus Power Available" = 250
          "USB Address" = 4
          "bMaxPacketSize0" = 64
          "iProduct" = 2
          "iSerialNumber" = 0
          "bDeviceClass" = 255
          "Built-In" = No
          "locationID" = 336592896
          "bDeviceSubClass" = 0
          "bcdUSB" = 512
          "USB Product Name" = "Gadget Serial v2.4"
          "PortNum" = 1
          "non-removable" = "no"
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "bDeviceProtocol" = 0
          "IOUserClientClass" = "IOUSBDeviceUserClientV2"
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "Device Speed" = 2
          "USB Vendor Name" = "Linux 3.2.35 with s3c-hsudc"
          "idVendor" = 1183
          "IOGeneralInterest" = "IOCommand is not serializable"
          "IOClassNameOverride" = "IOUSBDevice"
        }
```

May be need update firmware? But for what? At this moment I use latest from hantek official website

bikermikem, 2018-07-07:
I'm assuming you followed the build instructions re BREW.  The driver is built into the OpenHantek program as far as I know.  I'm not sure your model of scope is supported.  See screenshots.  The 5202P is not shown, just the 5200 & 5200A.  You might want to be careful, see:  https://www.eevblog.com/forum/testgear/dso5202p-autoset/.

[image]
[image]
[image]

WilixLead, 2018-07-08:
In github description I see "DSO52xx Series" in supported list, but looks like it is not true :(

bikermikem, 2018-07-08:
Well hopefully the coding gods will take note and try and fix this.

bikermikem, 2018-07-08:
You might want to look at https://github.com/OpenHantek/openhantek/blob/master/docs/adddevice.md