OpenHantek/openhantek #239: Compilation fails in Centos 7.5 with gcc 4.8.5

nbogiatzoglou, 2018-07-10:
When trying to compile openhantek using the latest Centos release, the compilation fails with the following error messages:

```
openhantek/openhantek/src/hantekdso/hantekdsocontrol.cpp:1226:24: error: missing binary operator before token "("
 #if __has_cpp_attribute(clang::fallthrough)
                        ^
openhantek/openhantek/src/hantekdso/hantekdsocontrol.cpp:1228:26: error: missing binary operator before token "("
 #elif __has_cpp_attribute(fallthrough)
                          ^
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/hantekdso/hantekdsocontrol.cpp.o] Error 1
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [all] Error 2
```

The compilation can be successfull and the program runs without any problems if the following statements are commented out in hantekdsocontrol.cpp:

```
//#if __has_cpp_attribute(clang::fallthrough)
//#define FALLTHROUGH [[clang::fallthrough]];
//#elif __has_cpp_attribute(fallthrough)
//#define FALLTHROUGH [[fallthrough]];
//#else
#define FALLTHROUGH
//#endif
```

FALLTHROUGH is only used at this point, so commenting out the if statements does not seem to affect anything.

closed 2018-07-10 by davidgraeff

Vascom, 2018-07-11:
@nbogiatzoglou Now you can install openhantek from rpmfusion repo http://koji.rpmfusion.org/koji/buildinfo?buildID=7846

nbogiatzoglou, 2018-07-16:
@Vascom Thanks for the info!

commit daa2dec6 (2018-07-10): Update hantekdsocontrol.cpp

Fixes #239