OpenHantek/openhantek #240: Legally unclear firmware source?

lemenkov, 2018-07-11:
Hello All!
We're trying to package OpenHantek driver for Fedora. Unfortunately current firmware sitoation is no-go for us.

Our question - have you tried to contact vendor to ask them to release firmware or get the permission for decomposition/dissection of their windows driver in order to get firmware?

davidgraeff, 2018-07-11:
Qingdao Hantek Electronic Co is the company and yes I tried to contact them like a year ago and never got an answer. In my country decomposition/dissection of public available software is allowed, which is why it got integrated into this repository.

The 6022xx firmware is a modified fx2lafw (LGPLv2).

In its current form this software is unfortunately not fitting for a linux distributions repository, I guess. Unless someone makes the effort and write another request to the mentioned company.

lemenkov, 2018-07-12:
@davidgraeff thanks for the clarification. We've got the picture.

closed 2018-07-12 by lemenkov

jnweiger, 2018-08-08:
If there is no response from Hantec, distributions would need DEB or RPM packages without the firmware.

Would it be possible to delay firmware extraction until runtime? If so, the software could, when first run, instruct the user "Firmware not found. Please run ... getfw.sh ... to fetch the firmware from http://www.hantek.com/Product/DSO200, then try again."

Vascom, 2018-08-08:
It would be good.
At that moment rpm package for Fedora and CentOS placed in rpmfusion nonfree repo.

davidgraeff, 2018-08-08:
Fetching the firmware and extracting from the Windows installer can be illegal in some countries as well, I guess. The only legal alternative of getting the permission would be a clean room firmware.

Ho-Ro, 2019-05-21:
@lemenkov & @Vascom 
[OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022) (supports only 6022 BE and BL) recently removed all nonfree (Hantek) firmware and uses free [firmware](https://github.com/Ho-Ro/Hantek6022API) based on the work of [Robert](https://github.com/rpcope1/Hantek6022API) and [Jochen](https://github.com/jhoenicke/Hantek6022API).

Vascom, 2019-05-21:
Now all firmwares in openhantek/res/firmware/ are free?

lemenkov, 2019-05-21:
> Now all firmwares in openhantek/res/firmware/ are free?

For me it looks so, but I am not a lawer :)

Ho-Ro, 2019-05-21:
Yes - OpenHantek6022/openhantek/res/firmware/ uses only GPLed free open source firmware from my repo where I added new features like variable calibration output frequency. This FW is based on the sigrok FW, which in turn is based on the work of Robert and Jochen as stated above.  
This FW replaces the old nonfree (Hantek) 6022 FW. As my fork doesn't support other models (2090, 2150, 2250, 5200) I also removed all other nonfree FW files that were extracted from the Hantek windows installers as well as the extraction code.
Only the branch [last_legacy_support](https://github.com/OpenHantek/OpenHantek6022/tree/last_legacy_support) and the old releases (up to v2.06) contain the old nonfree stuff. The upcoming v2.07 will be the first free release.

Vascom, 2019-05-21:
OK. So I can move openhantek rpm package from rpmfusion-nonfree repo to main fedora repo.

Ho-Ro, 2019-05-21:
yes, if you build against my master branch.

lemenkov, 2019-05-21:
@Ho-Ro cool news!
One last question (just for clarification) - does it means that using this new OSS firmware does not reduce a list of supported devices?

Ho-Ro, 2019-05-21:
My forked version OpenHantek6022 supports (as the name suggests) only the devices Hantek **6022BE** and **6022BL**. 
Reason: I own a 6022BE and develop and test accordingly with this device. With the help of other users I was able to define the exact HW differences between BE and BL and finally create a BL firmware. BE/BL now behave largely the same (except USB VID/PID) and can be handled by one program. 
Since I don't have another device (i.e. 2090, 2150, 2250, 5200), I decided to drop the support - I can't test. And as a logical consequence, I recently removed all related legacy program parts (many switch cases etc.).
In recent months I have received some feedback from users - problems with 6022BL that allowed me to understand the 6022BL HW and program an own FW version. 
Some macOS and windows problem feedback helped to improve support for "other-OS". 
But no 2xxx and 5xxx feedback - looks like these devices are pretty rare.

Vascom, 2019-06-06:
Now openhantek added to official Fedora GNU/Linux repo.