OpenHantek/openhantek #249: crash on startup in Virtualbox

suisseWalter, 2018-08-06:
I have the problem that Openhantek (2017-12 release) crashes on startup in my VirtualBox Ubuntu 18.04 with the following error: (it is a 6022BE)
"terminate called after throwing an instance of 'std::runtime_error *' Aborted (core dumped)"

according to this post:
https://github.com/OpenHantek/openhantek/issues/188
It could be fixed with the openhantek2 Branch. but I cant build this one. 

I'm a complete Linux newcommer therefore I don't really know how to find out more about this error. But i'm happy to try to help find the error in any way I can.

RayT76, 2019-04-11:
I had the same problem with my 6022BL, i also dual boot, but that has nothing to do with the problem.

I solved this problem by using https://codeload.github.com/OpenHantek/openhantek/legacy.zip/master

It configures as a 6022BE, and seems to work fine (only scope not logic tested at the mo.)

hope this helps

Ho-Ro, 2019-04-11:
You could try the latest version from my repo https://github.com/Ho-Ro/openhantek 
It has some improvements and a fully working custom FW for 6022BE and 6022BL.

RayT76, 2019-04-17:
I may give that a go, cheers Ho-Ro

Ho-Ro, 2019-04-17:
Hi RayT76

> the steps I used..
> sudo apt update && sudo apt install g++ cmake qttools5-dev qttools5-dev-tools libfftw3-dev binutils-dev libusb-1.0-0-dev libqt5opengl5-dev mesa-common-dev libgl1-mesa-dev libgles2-mesa-dev
> 
> open a terminal in the OpenHantek-openhantek-eb33325 folder.
> Then..
> 
> mkdir build
> cd build
> cmake ../
> make -j
> sudo make install
> 
> cd openhantek
> 
> plug the scope in and run with.
> 
> ./OpenHantek

For testing purpose you can omit the step `sudo make install` - only the udev rule must be in place (either in `/etc/udev/rules.d` or `/lib/udev/rules.d`) to enable the access rights for normal users on your computer - then you can execute the newly built file `openhantek/OpenHantek`. If everything works fine I would recommend to build a *.deb (as well as *.rpm and *.tar.gz) package file with `sudo make package` and install it via `sudo apt install ./openhantek_*_amd64.deb`. Or put it into ~/bin (this works at least on debian systems, create the directory if it doesn't exist already) if you want to test the new version only for yourself.

Have fun!