OpenHantek/openhantek #251: No record length option

blurrpp, 2018-08-08:
Hi i just downloaded and compiled openhantek. Works great for my hantek 6022BE but i completly can't record any data. Field "record length" is empty (no options available), i can save only short sample in CSV. Is thare any bug or there is something wrong in my system ?

dendvz, 2018-08-09:
The record length for 6022BE is fixed at 10KS (SW trigger turns it into 8.192kS). Drop-down box is completely disabled. What are the Samplerate and Timebase settings?

blurrpp, 2018-08-09:
Thanks for answer, it doesn't work with any Samplerate and Timebase. I've made some rough workaround I just save all mesured data to binary file and read it from this file with my simple program (file name I puted in command line of OpenHantek), maybe if i will understand better code i will try to make it works normaly.