OpenHantek/openhantek #253: Hantek 6022BE: extract firmware fails

waljoh, 2018-08-09:
Have spent three or more days attempting to get 6033BE up on Linux/Ubuntu, trying every branch and variant of the OpenHantek project. 
The common point of failure with every version of 'extractfw' is that it fails returning zero offset, e.g
"Symbol _firmware found (offset 0xffffffffffff7900, 0 bytes)
Can't get frimware contents: Bad value"
Tried about 5 different branches of openhantek, drivers from Hantek, the CD, different h/w, and upgraded from Ubuntu 16.04 to 18.04 - all with same or similar results. This is not new, many reports, but no solutions. 
openhantek-2017-12 is the nearest I have got. I have a red light flashing on the 6022BE, and the firmware software recognises the device but fails to download (nothing available to download). The program simply crashes immediately with a core dump. Another version (don't ask which) loads and runs, but does not see the scope.
Stuck and frustrated, but surely someone knows the answer?

alxpettit, 2018-08-16:
Wait, why was this closed!? I want to know more about this `extractfw` nonsense! I have a scope I'd like to try to add support for. What happened?

waljoh, 2018-08-17:
Hello Alexandria,

I closed it as I finally found something that worked..  for me at any rate.

I suppose that the symbols that extractfw looks for have been removed from the Hantek Windows drivers, as every version of extractfw, its predecessors or derivatives, that I found failed in exactly the same way, resulting in no file being created.

However the branch of OpenHantek that you get from downloading the .zip (and no doubt the .tar) from the header page (http://www.openhantek.org) will work, (while the download from GitHub linked on the same page did not help me. ) This is https://codeload.github.com/OpenHantek/openhantek/legacy.zip/master resulting in a file OpenHantek-openhantek-7862387(2).tar.gz.

If you extract the files from this, you will find the rules file under the firmware subdirectory, and under openhantek/res/firmware are the necessary firmware and loader files for the supported DSO's. However if you follow the build instructions and "sudo make install" the software manages to put these files in the correct places.  You do not need extractfw and its frustrations. The usb connects, and when the software is loaded and starts, it checks for the firmware on the device, and if not present it will install the firmware. A red light flashes, and turns green when operational on the 6022BE. This remains in situ until the device is switched off or disconnected.

I now have had the system built and running on a recent Dell Vostro running under Ubuntu 18.04 and also on an old Dell 32 bit laptop running Mint 19.04, but with the latter it took a while to get the necessary software downloaded, (I was fumbling about with Graphics software that I did not understand) and it needs to be configured to run with GLES ( command: OpenHantek --useGLES) in order to work. 

In order to assemble the necessary software to build, the most helpful source is not in the documentation supplied, but its equivalent in a different version (I am not sure of the provenance, I downloaded 10 or more different versions in the process) where the specific installation instructions are given for linux (and Mac and Windows) - attached. 

I had a nightmare and must delete the 20 or so downloads that I explored.. some being duplicates. I rejected this option early on as I had problems finding the right software and moved on. It may be OK for the initiated, but someone needs to have a good purge of the stables.

ATB

John
[readme_md.txt](https://github.com/OpenHantek/openhantek/files/2296747/readme_md.txt)

bikermikem, 2018-08-17:
I have it running on a pi 3.

alxpettit, 2018-08-18:
I do need extractfw because I wanted to try to extract firmware for my DSO5202P. :(

But I am glad you got a solution that works for you, at least! Best wishes.

kschoelpple, 2018-08-19:
Hi,
I also experienced same problem with red LED flashing, also message: "DSO-6022BE:Firmware upload failed"  after clicking on that I get also: "A device was found, but the firmware upload seem to have failed or the connection could not be established:"
I am running KUbuntu 18.04.1 
Kernel 4.15.0-32
Hantek 6022BE

Also tried different download files and noticed difference in firmware folders. On those downloads I get " Policy "CMP0072" is not known to this version of CMake."
After executing command cmake ../

waljoh, 2018-08-21:
Kschoelpie

Without understanding the nuts and bolts, the red light flashing signifies that something has been downloaded to the device to initialise it.. a good sign, but if the process does not complete not much use. 

If you follow the instructions I posted above and use the download version, you should get something to work. 

I had the Policy CMP0072 problem as well, but don't remember how I overcame it . At one point I commented out the line to get around it.

kschoelpple, 2018-08-22:
Hi,
after spending a few hours on it I got finally a result similar to waljoh:
-Connect Hantek device
- Run OpenHantek software (latest release)
- Red light starts flashing but actual software won't run but message "DSO-6022BE:Firmware upload failed" pops up - acknowledge this message by clicking on "connect to first device"
-Now click on a earlier version of OpenHantek (I believe a 2016 version)
Now everything works! 
Only downside is the initial view size is to large and screen size can not be adjusted until one of the settings windows on right hand side is minimized. (my screen is 1920x1080) 
Then screen can be adjusted to suit and all functions are accessible and working.
Hope this workaround may help others as well.

suisseWalter, 2018-11-11:
thank alot for this post, it worked well for me. the policy CMP0072 problem exsisted aswell. I simply solved it by commenting the 4th line in the cmakeLists.list (cmake_policy(SET CMP0072 NEW)) and then it compiled flawless and works great now. thanks alot

jsaada, 2018-11-16:
please someone help to post a step by step for the full process ?

i only can compile the version "openhantek/archive/2017-12.zip" on my ubuntu 18.04 but cannot extract the firmware

```
cd hex/
wget http://www.hantek.com/down.aspx?url=http%3a%2f%2fwww.hantek.com%2fProduct%2fHantek6000%2fHT6022_Driver.zip -O HT6022_Driver.zip
unzip HT6022_Driver.zip && rm *zip
cd HT6022BE_Driver/
mv *Win10 win10
rm *pdf
mv *xp* win7xp
cd ../..
for f in $(find hex -name *.sys); do 
echo "$f";
./build/FirmwareExtractor $f
done;
```

> Section .data found (starting at 0x2780, 8 bytes)
Symbol _firmware found (offset 0xffffffffffffd880, 0 bytes)
Symbol _loader found (offset 0xffffffffffffd880, 8 bytes)
Can't get firmware contents: Bad value
hex/HT6022BE_Driver/win10/Hantek6022BEIA641.sys
bfd_check_format: File format not recognized

i guess i need to compile / install the last version ( https://github.com/OpenHantek/openhantek/archive/openhantek2.zip ? ) but here is my output i'm stucked with

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt53DCore" with
  any of the following names:

    Qt53DCoreConfig.cmake
    qt53dcore-config.cmake

  Add the installation prefix of "Qt53DCore" to CMAKE_PREFIX_PATH or set
  "Qt53DCore_DIR" to a directory containing one of the above files.  If
  "Qt53DCore" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  openhantek/CMakeLists.txt:3 (find_package)
```

suisseWalter, 2018-11-16:
so you get the same window as waljoh in the first post here? and it then crashes on the second launch(....cpu core dumped....)?

jsaada, 2018-11-16:
i've got the core dumped you are talking about but i retried from scratch (see full procedure just above ) and after a reboot now 
EVERYTHING IS WORKING  \o/

mwette, 2019-03-23:
Hi All,

I have a DSO5072 on Ubuntu 18.04 and tried to run the FirmwareExtractor.  I got
```
mwette$ ./FirmwareExtractor dso5kp_func_dso5072p20190125.up 
bfd_check_format: File format not recognized
```
Then tried
```
mwette$ file dso5kp_func_dso5072p20190125.up 
dso5kp_func_dso5072p20190125.up: GPG symmetrically encrypted data (CAST5 cipher)
```
to deduce that it's not working because the firmware file is encrypted.  Maybe a change in firmware files.  I have installed the software but get "scope not support" message.   

I just get ``OpenHantek did not find any compatible devices.``  I'll dig a little more.