OpenHantek/openhantek #257: Feature Request - trace line weighting

LameLefty, 2018-08-28:
I'm not familiar at all with Qt, and I'm not entirely sure how widespread this issue might be, but I wonder if it is possible - and how difficult it would be - to enable some degree of line-weight or pixel-size control for trace displays? In my case, my main PC is probably an edge case - it's primarily a heavyweight gaming and productivity laptop with a native-resolution 4K display and a reasonably powerful graphics card to power it. As a result, the traces on the display tend to be fairly faint and more difficult to see than perhaps they could be.

A similar issue - though less serious - impacts the visibility of the graph markers and x-y axes of the display, though at least the color picker lets you increase or decrease the alpha-channel transparency to help. The lines and grid markers are still too light for my tastes but it's better than nothing.

Anyway, curious if anyone else feels this way or has a work-around? I've figured out how to compile OH under Windows, so I'm happy to test anyone's suggest code for a fix or modification.

stanleypa, 2019-01-09:
+1 - My laptop has a 4K display and the trace is hard to see at just 1 pixel width. Great project BTW - can finally use my 6022 on linux.

Anyeos, 2020-06-26:
I have no 4K but in 2K it is sometimes still difficult to see. Try changing the colors to ones that give more contrast. 

Another workaround is that you can use the system magnifier (the screen magnifier that is on your system, not related with this software). The magnifier normally is located on the accessibility section.

Ho-Ro, 2020-06-27:
Qt GL implementation doesn't support the OpenGL function `glLineWidth()`, search e.g. stackoverflow.
Also please see #277