OpenHantek/openhantek #262: Openhantek compiling fails due to libusb

szg0000, 2018-09-19:
```
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/main.cpp.o: in function `main':
/home/user/openhantek/openhantek/src/main.cpp:123: undefined reference to `libusb_init'
/usr/bin/ld: /home/user/openhantek/openhantek/src/main.cpp:132: undefined reference to `libusb_exit'
/usr/bin/ld: /home/user/openhantek/openhantek/src/main.cpp:210: undefined reference to `libusb_exit'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/selectdevice/rawdeviceslistmodel.cpp.o: in function `readUSBdescriptor(libusb_device_handle*, unsigned char)':
/home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:35: undefined reference to `libusb_get_string_descriptor_ascii'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/selectdevice/rawdeviceslistmodel.cpp.o: in function `RawDevicesListModel::updateDeviceList()':
/home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:49: undefined reference to `libusb_get_device_list'
/usr/bin/ld: /home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:57: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: /home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:62: undefined reference to `libusb_open'
/usr/bin/ld: /home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:71: undefined reference to `libusb_close'
/usr/bin/ld: /home/user/openhantek/openhantek/src/selectdevice/rawdeviceslistmodel.cpp:77: undefined reference to `libusb_free_device_list'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/finddevices.cpp.o: in function `FindDevices::updateDeviceList()':
/home/user/openhantek/openhantek/src/usb/finddevices.cpp:22: undefined reference to `libusb_get_device_list'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/finddevices.cpp:34: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/finddevices.cpp:65: undefined reference to `libusb_free_device_list'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/uploadFirmware.cpp.o: in function `UploadFirmware::startUpload(USBDevice*)':
/home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:23: undefined reference to `libusb_open'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:42: undefined reference to `libusb_set_auto_detach_kernel_driver'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:43: undefined reference to `libusb_claim_interface'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:69: undefined reference to `libusb_release_interface'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:70: undefined reference to `libusb_close'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:64: undefined reference to `libusb_error_name'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:65: undefined reference to `libusb_release_interface'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:46: undefined reference to `libusb_close'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:45: undefined reference to `libusb_error_name'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/uploadFirmware.cpp:54: undefined reference to `libusb_error_name'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::computeUSBdeviceID(libusb_device*)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:50: undefined reference to `libusb_get_port_numbers'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::USBDevice(DSOModel*, libusb_device*, unsigned int)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:56: undefined reference to `libusb_ref_device'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:57: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::claimInterface(libusb_interface_descriptor const*, int, int)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:105: undefined reference to `libusb_claim_interface'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::disconnectFromDevice()':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:134: undefined reference to `libusb_close'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:138: undefined reference to `libusb_unref_device'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:130: undefined reference to `libusb_release_interface'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::connectDevice(QString&)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:65: undefined reference to `libusb_open'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:75: undefined reference to `libusb_get_config_descriptor'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:89: undefined reference to `libusb_free_config_descriptor'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:94: undefined reference to `libusb_get_bus_number'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:95: undefined reference to `libusb_get_device_address'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/usbdevice.cpp:89: undefined reference to `libusb_free_config_descriptor'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::bulkTransfer(unsigned char, unsigned char const*, unsigned int, int, unsigned int)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:157: undefined reference to `libusb_bulk_transfer'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o: in function `USBDevice::controlTransfer(unsigned char, unsigned char, unsigned char*, unsigned int, int, int, int)':
/home/user/openhantek/openhantek/src/usb/usbdevice.cpp:190: undefined reference to `libusb_control_transfer'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/ezusb.cpp.o: in function `ezusb_write(libusb_device_handle*, char const*, unsigned char, unsigned int, unsigned char const*, unsigned long)':
/home/user/openhantek/openhantek/src/usb/ezusb.cpp:124: undefined reference to `libusb_control_transfer'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/ezusb.cpp:128: undefined reference to `libusb_error_name'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/ezusb.cpp.o: in function `ezusb_cpucs(libusb_device_handle*, unsigned int, bool)':
/home/user/openhantek/openhantek/src/usb/ezusb.cpp:163: undefined reference to `libusb_control_transfer'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/ezusb.cpp:163: undefined reference to `libusb_control_transfer'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/ezusb.cpp:170: undefined reference to `libusb_error_name'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/ezusb.cpp.o: in function `ezusb_read(libusb_device_handle*, char const*, unsigned char, unsigned int, unsigned char const*, unsigned long) [clone .constprop.3]':
/home/user/openhantek/openhantek/src/usb/ezusb.cpp:143: undefined reference to `libusb_control_transfer'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/ezusb.cpp:147: undefined reference to `libusb_error_name'
/usr/bin/ld: CMakeFiles/OpenHantek.dir/src/usb/ezusb.cpp.o: in function `ezusb_fx3_jump':
/home/user/openhantek/openhantek/src/usb/ezusb.cpp:186: undefined reference to `libusb_control_transfer'
/usr/bin/ld: /home/user/openhantek/openhantek/src/usb/ezusb.cpp:192: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/build.make:1062: openhantek/OpenHantek] Error 1
make[1]: *** [CMakeFiles/Makefile2:218: openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

davidgraeff, 2018-09-20:
Have you followed the build instructions especially installing the dependencies?

szg0000, 2018-09-23:
Thanks, it works.

closed 2018-09-23 by szg0000