OpenHantek/openhantek #271: recommendations for tablets or other devices yo run this software on.

lee3836hester, 2018-10-11:
For a person of limited means, what would be a minimally sufficient piece of hardware to get good functionality?

LameLefty, 2018-10-12:
I can't speak to tablets, but as I've recently determined in #258, a Raspberry Pi 3 (or 3 Plus) will run OpenHantek perfectly. That's currently about $35 for the Pi, $10 for a PSU, $10 for a microSD card, $10 for a case,  plus whatever you choose to use for a keyboard, monitor and mouse.  An even less expensive RPi Zero W will also "work," but performance is frankly too sluggish to make it worthwhile. Spend $25 more for something usuable.

davidgraeff, 2018-10-12:
Perfect answer. Minimum requirements are probably about 500mb ram, OpenGL 3 or OpenGL es 2 capable and around 1ghz cpu

closed 2018-10-12 by davidgraeff