OpenHantek/openhantek #272: Ubuntu 18.04 cmake build failure (pthread)

unrealsolver, 2018-10-17:
Probably, should be linked with `-lpthread` option, not `-lpthreadS`. Have no idea how tell it to cmake (setting ld option for Threads module won't work). Env: Ubuntu 18.04 amd64
Output:
```
$ cmake ../
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:4 (cmake_policy):
  Policy "CMP0072" is not known to this version of CMake.

-- Version: master/b8b43ad
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found libfftw3:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/x86_64-linux-gnu/libfftw3.so
-- Configuring incomplete, errors occurred!
See also "/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeOutput.log".
See also "/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeError.log".
```
Error log:
```
tail -n 50 CMakeFiles/CMakeError.log 
make[1]: Entering directory '/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_651da.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_651da.dir/CheckSymbolExists.c.o   -c /home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_651da
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_651da.dir/link.txt --verbose=1
/usr/bin/cc      CMakeFiles/cmTC_651da.dir/CheckSymbolExists.c.o  -o cmTC_651da 
CMakeFiles/cmTC_651da.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_651da.dir/build.make:97: recipe for target 'cmTC_651da' failed
make[1]: *** [cmTC_651da] Error 1
make[1]: Leaving directory '/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_651da/fast' failed
make: *** [cmTC_651da/fast] Error 2

File /home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e48d1/fast"
/usr/bin/make -f CMakeFiles/cmTC_e48d1.dir/build.make CMakeFiles/cmTC_e48d1.dir/build
make[1]: Entering directory '/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_e48d1.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_e48d1.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_e48d1
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e48d1.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_e48d1.dir/CheckFunctionExists.c.o  -o cmTC_e48d1 -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_e48d1.dir/build.make:97: recipe for target 'cmTC_e48d1' failed
make[1]: *** [cmTC_e48d1] Error 1
make[1]: Leaving directory '/home/ruslan/dev/openhantek/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_e48d1/fast' failed
make: *** [cmTC_e48d1/fast] Error 2
```

unrealsolver, 2018-10-17:
Did a trick with removing pthreads lines away from openhantel/CMakeLists.txt, also had to apply this patch https://github.com/OpenHantek/openhantek/pull/269

StephanRichter, 2018-10-25:
This is a duplicate of https://github.com/OpenHantek/openhantek/issues/265