OpenHantek/openhantek #273: Measurement immediately stops in Single Trigger mode without event

abusse, 2018-10-18:
Hi everyone,

I just built Openhantek from the latest commit and have in issue with the trigger in single mode. Instead of stopping at a rising or falling edge, the measurement stops almost immediately without the measurement getting anywhere close to the set trigger value. I'm using a 6022BL, if this is important.

StephanRichter, 2018-10-25:
Today my Device has arrived. I Built the sources and played around. And I can confirm this observation.

matthiesenj, 2018-12-07:
I also have this problem.. Pressing the play button, it momentarily shows the pause icon (||), then immediately returns to showing the play (|>) icon. OpenHantek 30.09.2018_na on windows 10 64 bit.

abc123me, 2019-01-05:
Same issue here on linux

Zbysekz, 2019-01-05:
Same issue here, ubuntu 18, Hantek6022BL

claus007, 2020-09-17:
Same issue here, Hantek6022BL
Is there still development going on ?

Ho-Ro, 2020-09-17:
Please read #277