OpenHantek/openhantek #276: Trigger slope dropdown shows question marks

matthiesenj, 2018-12-07:
OpenHantek 30.09.2018_na on Windows 10 64bit (from https://ci.appveyor.com/api/buildjobs/5000b2hr0lsageq4/artifacts/build_x64%2Fopenhantek%2FOpenHantek-x64-Release-b33.zip).
The slope dropdown has two items, both being a question mark. Looking at various screenshots, this seems to be wrong.
[image]

LameLefty, 2018-12-08:
I have the same issue when I use Win10 in a version compiled myself from source. The issue does NOT occur with Ubuntu in VirtualBox on my Win10 machine, nor in Raspbian on my RPi3. I'm guessing its an issue with the glyphs in the versions of Qt for the two different platforms (Windows v. Linux).

AndreasDerFuchs, 2020-05-03:
Issue persists on Win10 with OpenHantec 14.01.2020_na.
Not only the selections are just two questionmarks, furthermore the trigger does not work, too bad.

Ho-Ro, 2020-05-03:
> OpenHantek 14.01.2020_na

You're talking about this repo? Where did you get the version above? Do you know #277?
If you're interested in a maintained version, please have a look at [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022).

AndreasDerFuchs, 2020-05-03:
> > OpenHantek 14.01.2020_na
> 
> You're talking about this repo? Where did you get the version above? Do you know #277?
> If you're interested in a maintained version, please have a look at [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022).

I had that version from here:
https://github.com/OpenHantek/openhantek
--> followed the Link "Download Windows build" to
https://ci.appveyor.com/project/openhantek/openhantek/branch/master/artifacts
--> clicked on "Current build" -->  "Artifacts" and downloaded the file:
[build_x64\openhantek\OpenHantek-x64-Release-b119.zip](https://ci.appveyor.com/api/buildjobs/ig3pb0hrxb0otoo6/artifacts/build_x64%2Fopenhantek%2FOpenHantek-x64-Release-b119.zip)

But OK, now I see:
http://openhantek.org/
--> https://github.com/OpenHantek/OpenHantek6022
--> https://github.com/OpenHantek/OpenHantek6022/releases
--> The package [OpenHantek-Win-x64-Release-b259.zip](https://github.com/OpenHantek/OpenHantek6022/releases/download/3.0.4b/OpenHantek-Win-x64-Release-b259.zip) (Version 3.0.4b) is far better and apparently the issue is fixed in this one.
Thanks for the hint!