OpenHantek/openhantek #277: Maintainership

davidgraeff, 2018-12-07:
This software is looking for a new maintainer.
Until someone has been found it counts as abandoned unfortunately.

I will add a notice to the readme file in a few weeks.

Damijonas, 2019-02-08:
What does it takes? Maybe I'm up for the challenge. It's time to give back to the community.

Damijonas, 2019-02-08:
A whole lotta love 😃

Ho-Ro, 2020-06-29:
There is a maintained fork of this project - but only for Hantek6022 devices:
https://github.com/OpenHantek/OpenHantek6022

alfredopalhares, 2022-05-16:
Hello, I just got an Hantek Hantek 6074BC, and I might apply myself as maintainer once I get some familiarity with the codebase.

UpDownSide, 2023-03-29:
> Hello, I just got an Hantek Hantek 6074BC, and I might apply myself as maintainer once I get some familiarity with the codebase.

Have manage to add support?

sixtyfive, 2024-04-10:
Anyone still have a binary and could upload it somewhere?

ashevche, 2025-01-23:
Wouldn't it be better to move on and add a support to SigRok instead?