OpenHantek/openhantek #279: DSO-6022BL, firmware upload fail

nanogennari, 2018-12-11:
When i try to open OpenHantek with my DSO-6022BL connected the message "DSO-6022LE: firmware upload failed" appears and it wont connect. If i try to open it again without disconecting the USB OpenHantek just crashes with the message on the terminal: "terminate called after throwing an instance of 'std::runtime_error*'"

I have the udev rules instaled. And yes my device is a 6022BL and the message says 6022LE. And i've already tested it with the official hantek software to check if the hardware is working properly.

suisseWalter, 2018-12-12:
Have you checked out the issues 253 here on the github? it seams to be the same error. just build the one from the website and change the cmakefile.

nanogennari, 2018-12-12:
Thanks, it solves my problem also.

closed 2018-12-12 by nanogennari