OpenHantek/openhantek #284: Voltcraft DSO-2020USB

apparentVJ, 2019-01-21:
Hello,

I am trying to get the Voltcraft DSO-2020USB running, as it is really the same as the Hantek 6022BE.
the only thing is that I can't find a firmware for the 2020. 

Do anybody managed to get it working on OSX or Linux ? Any help is appreciated.

apparentVJ, 2019-01-24:
Bus 001 Device 006: ID 04b4:2020 Cypress Semiconductor Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x04b4 Cypress Semiconductor Corp.
  idProduct          0x2020 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          171
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 152 lines cut ...]
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

closed 2019-01-25 by apparentVJ

richardricci, 2019-08-30:
Hello!
Where can I find the code for the DSO 2020 mac? I just compiled and it's not showing up.
Thank you!
Richard

apparentVJ, 2019-08-30:
Hello there,
If you are looking to make the DS 2020 to work on any platform. you should look to change some files. it's not that hard at all.
this link is my repo for the dso2020 which worked for me on mac and raspberry pi.
https://gitlab.com/floatLab/openhantekdso2020

I have just copied another model file and some minor tweaks here and there, I can't recall right now exactly. I hope it is helpful and you can make it work.
If you still need help, you can write me ohere on this email.

All the best,Norbert

Ho-Ro, 2019-08-30:
Hi, 
This repo is unmaintained but I've just implemented the DSO-2020 in my repo OpenHantek6022 [just around the corner](https://github.com/OpenHantek/OpenHantek6022) -> https://github.com/OpenHantek/OpenHantek6022/commit/0f8a13655ed9f63b92c26fee11fb2af3c234c59c.
Please check and report any issues.

Have fun
Martin

richardricci, 2019-09-02:
I was able to build it without problems on Mac OS Sierra, thank you!!
Richard