OpenHantek/openhantek #288: Cannot seem to install the required drivers on Windows

neilbalch, 2019-02-16:
I configured the driver as described [here:](https://github.com/OpenHantek/openhantek/blob/master/docs/build.md#windows)

```
...
; =====================================================
; ========= START USER CONFIGURABLE SECTION ===========
; =====================================================

;DeviceName = "Hantek 6022BL"
DeviceName = "Hantek 6022BL"
; Make sure "VID_" and "PID_" are always part of the strings below
;VendorID = "VID_04B5"
;ProductID = "PID_602A"
VendorID = "VID_04B5"
ProductID = "PID_602A"
DeviceGUID = "{d35924d6-3e16-4a9e-9782-5524a4b79bac}"
DeviceClassGUID = "{79046022-7904-7904-1980-198007056022}"
; Date MUST be in MM/DD/YYYY format
Date = "02/16/2019"

; =====================================================
; ========== END USER CONFIGURABLE SECTION ============
; =====================================================
...
```

However, whenever I try to install the driver to my system, Windows fires back saying that it's an invalid driver and that the ones it currently has are newer.

LameLefty, 2019-02-24:
Have you tried the basics? Disconnect the Hantek from your PC. Open the Windows Device Manager and find any reference to the device and delete them all. Reboot. Now when you connect the device, point Windows at the your new file for the device driver.

That said, I had very mixed results with this kind of approach. To use OpenHantek (or sigrok) under Windows with my Hantek 6022BE, the only way I could get it to work reliably was to use zadig (https://zadig.akeo.ie/) to install a generic USB driver for the device instead.

neilbalch, 2019-02-24:
@LameLefty Yes, how?