OpenHantek/openhantek #291: Suggestion for successful first time users.

superthaix, 2019-03-02:
Hi everyone and thanks for making OpenHantek available to everyone. However I wanted to report my rough start.

I was excited to purchase a 6022BL and use OpenHantek (LInux) only to later bang my head against the wall for 2 days with the infamous "firmware cannot be dowloaded" error. The install went perfectly fine with no error however the error message persisted. 

The solution, was to download and reinstall the version on the GitHub website instead of the version available via the openhantek.org. It seems like the versions on openhantek.org are old and or outdated, not sure, but they didn't work, although they installed with no error.

For new users, it's not apparent that there are different version between GitHub and hopenhantek.org. 

You may want to add a notice on both sites suggesting that the version on GitHub may be the best option to start with.

Thanks
Brent

Ho-Ro, 2019-03-02:
Hi Brent, 
This github repo is unmaintained at the moment :( If you want to live on the bleeding edge move over to my fork https://github.com/Ho-Ro/openhantek where I do some further development (only 6022 on linux is tested due to missing other devices and systems). Latest binaries for linux, osx and win are available also but I would propose to compile your own software. Feel free to comment and propose further improvement (pull requests are welcome).
Martin

superthaix, 2019-03-04:
Hi Martin.

Thanks for the explanation, advise and invitation. I will surely give your fork a try. I'm planning to set up a semi permanent mini desktop pc (raspberry pi) to use openhantek with. I look forward to your 'bleeding edge' fork.

thanks
B

⁣Sent from BlueMail ​

Ho-Ro, 2019-03-04:
please check also this info: https://github.com/Ho-Ro/openhantek/issues/2

voltagex, 2019-04-24:
@Ho-Ro my suggestion would be try to get in contact with one of the repository owners and see if you're able to be given contributor access.

voltagex, 2019-04-24:
@davidgraeff?

davidgraeff, 2019-04-24:
Contributor access is no problem. I would also rather see development happening in one central place.

If someone want's to tackle binary generating scripts, that's also very welcome

Ho-Ro, 2019-04-24:
@davidgraeff 
David, where do you see my *"OpenHantek6022"* approach in the overall frame of your OpenHantek project? 

I put a lot of effort in this sidestep to get the most out of *my* 6022BE without doing to much harm to all *"non-6022"* devices. But I didn't get any input from users of the non-6022 devices if the program is still working as expected. So I thought already about creating a branch without all 2xxx and 5xxx support.

Two more points:
- The firmware for 6022BE/BL is now based on open source. Due to the response of two test users I was also able to create an OS firmware for 6022BL - these two devices behave now similar. My (future) plan is to implement isochronous transfer into OpenHantek to minimise overruns - the new FM already supports iso.

- Another user guided me through the hassles to get OH running flawlessly under macos - OH is now available from macports.

Martin

davidgraeff, 2019-04-24:
openhantek was originally just a project for the 5xxx devices and got extended over time... and not very well. It is badly designed internally. I tried two times to refactor it, but I do not own those old devices and cannot test properly.

So yes I do see advantages of having a "OpenHantek6022". But I also ask myself, why not use sigrok?

Ho-Ro, 2019-04-24:
Hello David, 
I totally agree with you that OpenHantek development should happen here in one central place. What do you think how the new structure should look, a historical branch with all devices and the new OpenHantek6022 branch? And how do we transfer my fork *back home*?
Nevertheless the timing is very appropriate, I'm just at the state that my OH6022 is mostly complete - the next possible improvement steps (besides the ISO xfer) would require HW hacks on the scope, e.g. AC input coupling.
And regarding your question *"why not use sigrok?"* - I like the look and feel of a scope, pulseview looks more like a chart recorder - but sigrok is a perfect tool if you want to decode i2c or uart communication.

davidgraeff, 2019-04-24:
> pulseview looks more like a chart recorder

The advantage of sigrok is the strong decoupling of the user-interface from the decoding backend.

I'd say you move your repo over to the OpenHantek team and we'll "promote" your binary for that specific scope in the readme and on the webpage.

Ho-Ro, 2019-04-24:
Ok, I've renamed my openhantek to OpenHantek6022 and just tried to transfer it but get the error:
*OpenHantek already has a repository in the OpenHantek/openhantek network and You don’t have the permission to create repositories on OpenHantek* - ???

Ho-Ro, 2019-04-25:
Ok, transfer done: https://github.com/OpenHantek/OpenHantek6022

Vascom, 2019-04-25:
All development now will be there?

Ho-Ro, 2019-04-25:
Yes

Ho-Ro, 2019-09-03:
> I'd say you move your repo over to the OpenHantek team and we'll "promote" your binary for that specific scope in the readme and on the webpage.

@davidgraeff:
Hi David,
I was asked why OpenHantek6022 is not mentioned on openhantek.org, do you see a chance to give me edit access? I would have sent a pull-request if I had found the source in the structure of OpenHantek.
Martin

davidgraeff, 2019-09-03:
The website is a github pages generated page from the "docs" directory of the master branch.
Just make a PR and the website will be rerendered on merge.