OpenHantek/openhantek #296: Error in Windows version OpenGL

pldestefa, 2019-05-29:
Hi, I have the following error that I attach in the images, I am not a professional programmer, I do what I can, do you think I can solve it?

Thank you!
[image]
[image]

Ho-Ro, 2019-05-29:
Hi, you could try to start your openhantek with "openhantek --useGLES", see also #221. Or instead of using the unmaintained openhantek you can give my actively supported fork a try: https://github.com/OpenHantek/OpenHantek6022

pldestefa, 2019-05-29:
Ho-Ro, thank you very much!  With the command line it worked.  I will also try the other version that you recommended me.

closed 2019-05-29 by pldestefa