OpenHantek/openhantek #301: hantek 6022be connection failed win 10

seraphie21, 2019-07-07:
hey, recently i'm searching for replacement software for the hantek 6022be and i found openhantek, but everytime its always say connection failed, i've tried this on 4 diff pc with win 10 and always get the same problem

Ho-Ro, 2019-07-08:
openhantek ist unmaintained (#277) and deprecated. Pleased try my [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022). For windows you have to follow the driver installation procedure: https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#windows
As I don't use windows I checked it only very limited in a virtual win7 installation.

seraphie21, 2019-11-26:
hey, sorry for long reply, because i forget the email for this acc, anyways, thanks for your help, but is till cant make it work, and you mentioned that "2nd time for the scope with firmware uploaded", what's that mean and is it not included at zadig ?

Ho-Ro, 2019-11-26:
When you connect the scope, it starts without firmware and shows USB vendor / product ID 04B4/6022. When openhantek detects this device, the firmware is uploaded into the scope's RAM and the scope disconnects and automatically connects to USB, but this time (with FW in RAM) as USB vendor / product ID 04B5/6022. 
For Windows it looks like two different devices, both (with/without FW) use libusb for USB communication and therefore both must be handled by zadig - 1st time 04B4/6022, 2nd time 04B5/6022.
[Read more](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#windows).

seraphie21, 2019-11-27:
hey, i've already tried that, and i just read raxis13's success report, and apparently it just work with the build 68, but for the newer version it always failed, so yeah, thank you for your help and time, hopefully there's a new build for windows, thank you so much 😄 😄 😄

Ho-Ro, 2019-12-05:
@seraphie21 
Please close here as nobody will fix it for openhantek right now. Your concern is handled by this issue: 
[Hang/crash at device detection when OS != Linux (Was: Cant Install windows pre-built version) #38
](https://github.com/OpenHantek/OpenHantek6022/issues/38)

seraphie21, 2019-12-09:
@Ho-Ro okay, i'll close this, thanks for the help :)

closed 2019-12-09 by seraphie21