OpenHantek/openhantek #302: MSO5202

artiomn, 2019-08-15:
Is it possible to add support for the MSO5202?

Ho-Ro, 2019-08-15:
openhantek is unmaintained at the moment - but you can try it on your own.
The only supported scope is currently Hantek 6022BE/BL: https://github.com/OpenHantek/OpenHantek6022

artiomn, 2019-08-16:
Thanks. I have tried to add (by inheritance from DSOModel class), but the software can't connect to my device.

```cpp
#pragma once

#include "dsomodel.h"

class HantekDsoControl;
using namespace Hantek;

struct ModelMSO5202D : public DSOModel {
    static const int ID = 0x505a;
    ModelMSO5202D();
    void applyRequirements(HantekDsoControl* dsoControl) const override;
};
```

tangram67, 2021-01-16:
Hi to all.

I added the MSO5202D to the list of supported models:

static std::list<DSOModel> supportedModels =
    std::list<DSOModel>({DSOModel(MODEL_DSO2090, 0x04b5, 0x2090, 0x04b4, 0x2090, "dso2090x86", "DSO-2090"),
                         DSOModel(MODEL_DSO2090, 0x04b5, 0x2090, 0x04b4, 0x8613, "dso2090x86", "DSO-2090"),
                         DSOModel(MODEL_DSO2150, 0x04b5, 0x2150, 0x04b4, 0x2150, "dso2150x86", "DSO-2150"),
                         DSOModel(MODEL_DSO2250, 0x04b5, 0x2250, 0x04b4, 0x2250, "dso2250x86", "DSO-2250"),
                         DSOModel(MODEL_DSO5200, 0x04b5, 0x5200, 0x04b4, 0x5200, "dso5200x86", "DSO-5200"),
                         DSOModel(MODEL_DSO5200A, 0x04b5, 0x520a, 0x04b4, 0x520a, "dso5200ax86", "DSO-5200A"),
                         **DSOModel(MODEL_MSO5202D, 0x049f, 0x505a, 0x049f, 0x505a, "dso5202dx86", "DSO-5202D"),**
                         DSOModel(MODEL_DSO6022BE, 0x04b5, 0x6022, 0x04b4, 0x6022, "dso6022be", "DSO-6022BE"),
                         DSOModel(MODEL_DSO6022BE, 0x04b5, 0x602a, 0x04b4, 0x602a, "dso6022be", "DSO-6022LE")});
                         
Unfortunately that doesn't help much, because the firmware extraction tool does not find the firmware in the windows drivers for the MSO5202D:

BFD: hex/MSO5000D_Driver/dstusbamd64.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: hex/MSO5000D_Driver/dstusbamd64.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: hex/MSO5000D_Driver/dstusbamd64.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
BFD: hex/MSO5000D_Driver/dstusbamd64.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .pdata
Section .data found (starting at 0x1c80, 272 bytes)
Symbol _firmware found (offset 0xffffffffffffe380, 0 bytes)
Symbol _loader found (offset 0xffffffffffffe380, 272 bytes)
Can't get firmware contents: Bad value
bfd_check_format: File format not recognized
BFD: hex/MSO5000D_Driver/dstusbx86.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .text
BFD: hex/MSO5000D_Driver/dstusbx86.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .rdata
BFD: hex/MSO5000D_Driver/dstusbx86.sys: Warning: Ignoring section flag IMAGE_SCN_MEM_NOT_PAGED in section .data
Section .data found (starting at 0x1380, 8 bytes)
Symbol _firmware found (offset 0xffffffffffffec80, 0 bytes)
Symbol _loader found (offset 0xffffffffffffec80, 8 bytes)
Can't get firmware contents: Bad value

For now the model is shown on startup, but then an error message is displayed because of the missing firmware files.

[image]
[image]

artiomn, 2021-02-03:
> It would be great if someone can give a hint of how to overcome the missing firmware files?
> Maybe then the DSO part of the MSO5202D will work as a DSO52xxx type.....

Maybe contacting Hantek will help to get the firmware?
I'll try to write service@hantek.com...
It's better, than nothing.

tangram67, 2021-02-04:
I already tried to get in touch with the HANTEK service people to get a current firmware update file for the MSO5202D. There is no link for current firmware versions for this device on the HANTEK homepage. The reason to contact the HANTEK service was that I'm having an annoying boot error message that says that the device was not properly shut down. My idea was to fix this by a clean firmware update on the newly purchased and absolutely unmodified scope. So I asked the HANKTEK support for a current firmware file, but the response was less than zero, also no response in the eediscuss.com forum for weeks. Maybe the MSO5202D is already end of life, so no further updates will be available for this model.

But I found a firmware file MSO5202D_Firmware(3.2.35).zip from 2016 elsewhere in the internet. But unfortunately the gpg decyption is not working with the default password '111111, qpg complains about the wrong passphrase.

artiomn, 2021-02-07:
> There is no link for current firmware versions for this device on the HANTEK homepage. 

Yes, I've noticed this too.

> The reason to contact the HANTEK service was that I'm having an annoying boot error message that says that the device was not properly shut down.

Strange, but I haven't any errors. So, my firmware is different?

> So I asked the HANKTEK support for a current firmware file, but the response was less than zero

I've mailed them too, but they didn't response.

> Maybe the MSO5202D is already end of life, so no further updates will be available for this model.

But they sell this model [even on the official site](http://www.hantek.com/products/detail/10) and on the [Russian site](https://www.hantek.ru/products/mso5202d.html) too.

> also no response in the eediscuss.com forum for weeks

You can try to write directly to user Amy in the forum, [he's Hantek employee](https://www.eediscuss.com/forum.php?mod=viewthread&tid=9352&highlight=mso5202) and he must to know password and [has firmwares](https://www.eediscuss.com/forum.php?mod=viewthread&tid=9336&highlight=mso5202).
I'll try to write to his working e-mail:  caiyijun@hantek.com .

tangram67, 2021-02-07:
@artiomn
Thanks for the mail address. I will try to contact Amy. I hope that he will respond....

Regarding the firmware version:
I bought my device at the end of 2020 and the version installed is 3.2.35 (190514.0). Looks like the a version from 14.05.2019. So that already may be the latest version. But I hope when reinstalling the firmware the annoying boot message disappears. I costs only time at bootup....

artiomn, 2021-02-08:
My firmware version is older: 3.2.35 (151208.0). I've written them too. But they took a vacation: Chinese New Year.

tangram67, 2021-02-08:
Okay, let's see if we get some response...
But I'm sure, not within the next 2 weeks ;-)

Wedmer, 2021-02-15:
I have same 3.2.35, but it was built in December 2014.

tangram67, 2021-02-16:
Yes I also think the latest "version" is 3.2.35 build on the given date, e.g. 190514. I believe the date is checked when trying to update the device.

But the question is, will the firmware include the necessary files to start openhantek for a MSO5202?
Usually the "firmware" files for openhantek were derived from the driver files. So the idea was that these files were also include in the official firmware update files. To check this we are still in need of the password to decrypt the firmware update archive.

And maybe the firmware files must fit to the current version on the scope ;-)