OpenHantek/openhantek #316: DSO5102P support

wandiufo, 2020-05-28:
Dear developer

I have DS5102P oscilloscope. Will openHantek support for the model?
How to make support for the oscilloscope model?
Thank you for your help and information. 

Best regards,
Irwandi

Ho-Ro, 2020-05-28:
Hi Irwandi, Please read #277!

wandiufo, 2020-05-30:
I have made some modification to support DSO5102, and it detected,
[image]
when click ok
[image]
the error message appears:
wandi@wandi-asus:~/save/ee/osiloscope/git/openhantek/build$ ./openhantek/OpenHantek 
"Couldn't get channel level data from oscilloscope"
compile shaders
compile shaders
Sending control command b4 failed: Operation timed out
Getting capture state failed:  "Operation timed out"
Sending control command b4 failed: Operation timed out
Getting capture state failed:  "Operation timed out"
Sending control command b4 failed: Operation timed out
Getting capture state failed:  "Operation timed out"
Sending control command b4 failed: Operation timed out
Getting capture state failed:  "Operation timed out"
Sending control command b4 failed: Operation timed out
Getting capture state failed:  "Input/output error"
Sending control command b4 failed: No such device (it may have been disconnected)

I think this different command between DSO5200 and DSO5100,
Because DSO5100 cannot recognize the CONTROL_SETOFFSET = 0xb4. And my oscilloscope be come reboot after running the command. 
Could someone give me information the different command protocol between DSO5200 and DSO5100 because I will modified the file controlcode.h, bulkStructs.h,  hantekdso/hantekdsocontrol.cpp or other files?

Thank you for your help and collaboration. 

Best regards,
Irwandi

barrydegraaff, 2020-09-08:
What did you change to make it detect and show Ready for use?

barrydegraaff, 2020-09-08:
See also:
https://github.com/titos-carrasco/DSO5102P-Python
https://elinux.org/Das_Oszi_Protocol