OpenHantek/openhantek #322: DSO-2090 record length greyed out on box power up.

vfbl, 2020-10-05:
Openhantek 30.10.2020_e7e0ec7b built on 64bit Debian with DSO-2090 USB.

The record length was initially greyed out when the box was power up (i.e. fresh firmware).   The setting would become available after flipping samplerate to Max (100MS/s).

[image]

fjansson, 2021-04-26:
I have the same problem, also with a DSO-2090. Perhaps because the record length is wrong/not set I get a very short piece of signal trace, filling maybe 10% of the window horizontally. (I could not set the sample rate above 50MS/s, and I didn't manage to make the record length setting available). This is using the current master branch, same commit e7e0c7b8 as vfbl.

To be able to use my scope at all, I tried the 2011 OpenHantek on sourceforge, https://sourceforge.net/projects/openhantek/files/ .
That version still works (had to patch it by adding a few `(double)` inside `abs()` calls to get it to compile). Unfortunately the firmware handling in that version is much more tedious. I also briefly tried pulseview from sigrok, which is supposed to support the  DSO-2090, but it crashed.

fjansson, 2021-04-26:
I just discovered this fork by @albertoscotta, https://github.com/albertoscotta/openhantek/tree/my-master
where the problem has been fixed! Just worked out of the box for me.

Also otherwise there seems to be good development in that fork. @davidgraeff could it get some kind of official link from this repo to be more discoverable ?