OpenHantek/openhantek #325: Не записывается прошивка в Hantek6022BE при установке драйвера

pu6k1n, 2020-12-08:
При двухуровневой установке драйверов с использованием zadig-2.5 в Win 7 32. При первом запуске OpenHantek и zadig-2.5 успешно проходит установка, в диспетчере появляется Unknown Device #1 (Версия драйвера: 6.1.7600.16385; Цифровая подпись: USB\VID_04B4&PID_6022 (libwdi autogenerated)). Но при втором перезапуске программы OpenHantek она выдает LIBUS_ERROR_INVALID_PARAM и в диспетчере устройств не появляется DSO-6022BE (прошивка не заливается в сам прибор) 
[image]

В чем может быть проблема?

Ho-Ro, 2020-12-09:
?

Vascom, 2020-12-09:
@pu6k1n Use english language please.
Пиши на английском.

Ho-Ro, 2020-12-17:
You could try the [install with the driver](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#signed-winusb-driver-for-hantek-6022bebl) that is available in latest [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable).

BTW: you post about OpenHantek6022 in the wrong repo, this one (openhantek) is currently unmaintained.

drf5n, 2021-12-29:
See https://github.com/OpenHantek/OpenHantek6022 for more up-to-date fork.