OpenHantek/openhantek #329: (Wrong) vertical scale

spassignat, 2021-06-13:
Hi,

here is a screen shot of a 5v signal. It appears to be 2.5v.

[image]

Am I doing something wrong ?

Ho-Ro, 2021-06-13:
Did you read [this](#277)?

DJSigmann, 2023-01-18:
I'm also experiencing the same error. Couldn't fix it no matter what I tried.

Ho-Ro, 2023-01-18:
@DJSigmann Which device? Which kind of signal? 
If I look into my glass ball I see a 6022 :)
Why don't you use the [right program](https://github.com/OpenHantek)?

djsigmann, 2023-05-16:
> @djsigmann Which device? Which kind of signal? If I look into my glass ball I see a 6022 :) Why don't you use the [right program](https://github.com/OpenHantek)?

Even better!
I was using the [right program](https://github.com/OpenHantek/OpenHantek6022), I just went to the wrong repo to raise an issue.

I'll raise one in the correct repo if I notice this bug the next time I use the program.