OpenHantek/openhantek #332: windows bin?

agentpatience, 2021-09-26:
Can someone provide windows x64 binary, the link to get it is down? or the archive is deleted?

Ho-Ro, 2021-09-26:
Please see #277!
Do you want an *openhantek* binary because you have a scope other than Hantek DSO-6022BE or ..BL? Otherwise you should try [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022).

agentpatience, 2021-09-26:
I just have DSO-2090 and have no idea how to build from source. I just want to run the binary for it. I remember openhantek has a better FFT analysis where it plots the data? I'll try your fork. Thanks for the info.

agentpatience, 2021-09-26:
I tried your fork v.2.06 but I get connection failed. When I boot to windows and use the Hantek drivers everything is working.

I have attached a screenshot to show you where the program stops responding.
[image]

Ho-Ro, 2021-09-27:
Did you install the [correct USB driver](https://github.com/OpenHantek/OpenHantek6022/discussions/224#discussioncomment-1356319)?

agentpatience, 2021-09-27:
Yes, now it works fine. Thank you kindly!

agentpatience, 2021-09-27:
whoops spoke too soon... seems to be really laggy and not adjusting scope properly to see any waveform?
[image]

Ho-Ro, 2021-09-27:
> seems to be really laggy and not adjusting scope properly to see any waveform?

You should try to lower the sample rate, 50MS/s is quite high for half a second on the screen (i.e. 25 MS). But as I have only a 6022BE this is just a shot in the dark.

To answer your earlier questions:

> I just have DSO-2090 and have no idea how to build from source.

Please check [check the docs](https://github.com/OpenHantek/openhantek/blob/master/docs/build.md#windows) and have a look [how it is done](https://github.com/OpenHantek/openhantek/blob/master/appveyor.yml) by appveyor.

> I remember openhantek has a better FFT analysis where it plots the data?

Yes, but (all ?) early releases have a nasty FFT bug - it uses only the real part of the complex spectrum instead of calculating the absolute value sqrt(re² + im²):
````
v2.08
2019-05-24
Major changes from v2.07:
This release fixes the quite unstable spectrum display. The annoying amplitude jitter has gone.
````